opentx/opentx #6806: RSSI telemetry not manageable.  9XR (2.2.4)  OrangeRX(DSMX11) Jumper JP4in1 Multi-Protocol Module

gweep, 2019-09-18:
RSSI telemetry not manageable.  
Turnigy 9XR (2.2.4)
OrangeRX R620v2(DSMX11) 
Jumper JP4in1 Multi-Protocol Module

With telemetry setup, the RSSI displays on the main screen as "signal bars" above the box for the left stick.  It also is displayed in in small type as a number in the footer of every telemetry screen setup to display bar charts.  

However RSSI does not show up in the Model Setup/Telemetry screen as a manageable data selection.  A full page of data from the receiver is available in the Model Setup/Telemetry screen for the user to manage and display in the Model Setup/Display screens.  RSSI is not included.  

Clearly RSSI is available from the receiver and is processed by OpenTX.   Perhaps Spectrum elected to omit the data but the OrangeRX receiver sends it anyway?

It would be nice to be able to display RSSI in big numbers on the telemetry screens and write logs.

raphaelcoeffic, 2019-09-18:
Silly question: have you done the sensor discovery on the model setup -> telemetry page?

gweep, 2019-09-18:
Yes.  I even deleted them all and started over.

I also note that the Display screen offers to show 4 lines of data (numbers and bars) but only 3 lines are visible in the display.

kilrah, 2019-09-18:
Spektrum gear does not give an RSSI. You might have a TSSI in your sensors.

closed 2019-09-18 by kilrah

gweep, 2019-09-20:
Yes TRSS show up and is manageable.   

Getting back to RSSI; even though it is not expected... RSSI is received from the OrangeRX receiver and is displayed by openTX.   Hence it could be made available to manage and display.

kilrah, 2019-09-20:
What you see displayed is the TSSI.

gweep, 2019-09-20:
In fact it is not fed into the RSSI mechanism.    The TRSS values and the RssI values differ by over 30 to 40 dB.

kilrah, 2019-09-20:
Sorry it's not TRSS, it's based on Frame Loss A.

https://github.com/opentx/opentx/blob/2.3/radio/src/telemetry/spektrum.cpp#L321

gweep, 2019-09-24:
Thanks!  That clears it up.   It also explains why my RSSI does not change when I switch to low power for a range check.

Thanks

3djc, 2019-09-24:
Just as a side comment, low power has been disabled in recent multi builds, so unless you are compiling yourself and have re enable low power, it might not be available