opentx/opentx #6809: Toggle switches do not work an connectors P20 and P32

kaeuferrot, 2019-09-20:
I installed OpenTx RC4 on my Horus X12S because I read that additional switches are supported with 2.3. On my sticks, I have to additional toggle switches which were connected to P20 and P32 on motherboard. In the hardware menu I can see the switches SI and SJ and I configured them as toggle. The servo monitor shows me bars. But when toggling, nothing happens. Also the way of testing "There is a simple way to test if they work : go to SPECIAL FUNCTIONS screen, select an empty one, enter first column and enter, your cursor should blink. Move your switch, if it appears here, it is working correctly " 3djc described in the 2.3 discussion thread was not successful.

3djc, 2019-09-20:
Please test tomorrow’s nightly, some changes have been done to try to fix this.

kaeuferrot, 2019-09-22:
Unfortunately I am not able to download the nightly even I reinstalled the N94-companion. The BIN file I get is only RC4. It was installed when I opened the issue.

3djc, 2019-09-22:
You need to have companion set to download nigthlies (or an nightly companion of it doesn not allow you to select nightly)

kaeuferrot, 2019-09-22:
Great! It works now! Many thanks!

3djc, 2019-09-22:
👍 thanks for the test !

kaeuferrot, 2019-09-22:
I can see the switches toggling and they are recognized when moving them. Servo monitor also shows moves. I tried to implement a special function, when pressing the key, that some values were played. But nothing happens there. With SH as a switch it works. Normally the switch name is bolded when active. Here this does not happen.

3djc, 2019-09-22:
What type of switch did you connect and how did you declare it ?

kaeuferrot, 2019-09-22:
It is a toggle stick switch from RC-Technik. I declared it as toggle!

kaeuferrot, 2019-09-22:
Acting as a channel works!

3djc, 2019-09-23:
Please test in RC5

closed 2019-09-23 by 3djc

kaeuferrot, 2019-09-23:
Thank you, it works now!