opentx/opentx #6811: Companion: Highlight orphaned logical switches in red

notlebowski, 2019-09-21:
When changing complex model setups it can happen that some of the logical switches get "orphaned" meaning they are not referenced by anything (not SF nor other logical switches).

If such switches would be highlighted in Companion this would help to show errors or remnants.

I just changed the way I turn on/off logging in one of my setups.

This means that part of the former logic may or may not be addressed anymore.
I don't dare to just remove the logical switch because it could be referenced for some other purpose as well.
This means I need to reconstruct the logic of the whole model, which is not only time consuming, but would also make it easier to make a mistake.

Companion highlighting a referenced switch would help

elecpower, 2019-09-23:
Orphaned references can occur in a number of areas and implementing for logical switches is likely to spawn requests for other reference checks. Performing lots of cross checks would slow down model loading and refreshing after every field change. Also colour coding is significant dev work.
My vote would be for a model 'audit' function.

notlebowski, 2019-09-24:
It says "Companion"

Tygrys-1, 2019-09-29:
Is it possible to find this dependency in the LUA scripts?

notlebowski, 2019-09-30:
Its strength would come from having it all the time (in Companion)
If suddenly one of the switches turns red, you may be doing something wrong.

notlebowski, 2020-03-31:
I think that this feature would help with complex setups and would help to avoid mistakes when editing them in Companion.

notlebowski, 2020-03-31:
I am interested to find out what the view of devs are regarding this.

bigtwin, 2020-09-27:
To me this seems like a desirable feature. If it is implemented in Companion I would think this is surely sufficient and would not harm any computing ressources in the transmitter. 
I already stumbled across orphaned LS and was wondering why no mechansim throws a warning or a hint.

closed 2021-09-04 by stale[bot]