opentx/opentx #6814: 2.3 RC4 minus button in LUA triggers twice

lshems, 2019-09-21:
On taranis (didn't check others) the minus button triggers the LUA event twice. The plus functions correctly.

3djc, 2019-09-23:
What event are you monitoring ?

lshems, 2019-09-23:
The 'old' ones.

lshems, 2019-09-23:
The 'old' ones. I just run my scripts and get double response on minus button. So something has changed.

kilrah, 2019-09-24:
Haven't been able to reproduce, so if it's still an issue in RC5 we'll need more detail and affected code.

ExaflopS, 2019-10-02:
I have found, the minus button in lua script, now in release version 2.3.0 minus button makes one - and one + step by pressing once, so if I click -, the value remains same, only way is to hold - button to go to lower valies and then use + button to go to desired value. however it is impossible to get value 0, because it always jump to 1.

3djc, 2019-10-02:
I'm not seing that here at all. Here you can find scripts to test and new and old events : https://github.com/opentx/opentx/tree/2.3/radio/util/lua

Plus and minus seem to be working ok here

ExaflopS, 2019-10-02:
I Have tested Yours test scripts. + and minus worked good, but only lua_old_event_check.lua worked in full. In lua_event_test.lua only next and previous are working , enter, menu, next rept, previous rept, next page and previous page are not responding.

3djc, 2019-10-02:
Yes a fix for new events  is comming in tomorows nightly

lshems, 2019-11-01:
For one reason or another, in the Taranis there is now a EVT_DOWN_BREAK event triggered after the release of the minus. this wasn't the case on 2.2.4.

The EVT_DOWN_BREAK should only be triggered for horus type radios.

lshems, 2019-11-02:
I'll close this as the root cause is a clear bug. This is just the effect in my scripts.

closed 2019-11-02 by lshems