opentx/opentx #6819: Companion telemetry conversion 2.3.0RC5

elecpower, 2019-09-23:
Rechecked the conversion after #6798 fixed and noticed the following changes from 2.2.4 otx but not sure if a conversion issue or the way data will be displayed in 2.3.

Version 2.2.4
[image]
Version 2.3.0RC5
[image]

My 2.2.4 flashed X9D+ radio displays the same as the Companion 2.2.4 screenshot.

**General information**
 - Hardware Frsky X9D+
 - OpenTX Companion 2.3.0RC5 on Ubuntu 16.04

raphaelcoeffic, 2019-09-23:
The instance does now correspond to what is actually received. In 2.2.4, it was +1, so the instance was actually wrong and has now been corrected. So it is in fact fully intentional.

Concerning the `()` you can see behind the sensor name in sources, this would contain the name of the receiver in case you would select a specific one and would be using ACCESS. This is very similar to the way this is displayed on the radio.

elecpower, 2019-09-23:
@raphaelcoeffic thanks for the explanation. Just wanted to double check. I do not have a spare radio to test the firmware on so couldn't check what the radio displays in 2.3.0RC5 to make sure Companion matches for the Instances. Happy to close.

closed 2019-09-23 by elecpower