opentx/opentx #6826: Failsafe Not Set Warning shown when switching between models using Multi Module

MRC3742, 2019-09-25:
**Describe the bug**
When changing from a model using Multi that has a failsafe option set to a model using Multi with no failsafe option available the second model will start to load and stop with the Failsafe Not Set warning screen, requiring you to press any button to continue.
 
**To Reproduce**
1. Load first model with external Multi selected and a Protocol that has a Failsafe Mode Option, (Such as "FRSKY" "D16" or "FSky 2A" "PWM,IBUS").
2. Set Failsafe Mode to any option (Such as "Receiver").
3. Change to second model that already has Multi selected and using a Protocol with no Failsafe Mode available, (Such as "Hisky" "Std").
4. As this model loads it will trigger the Failsafe Not Set screen - Press any button to continue.
5. Once this second model is selected shutting down and restarting Taranis will not trigger this Failsafe Not Set warning screen.

6. Note: Also the first model will never trigger the Failsafe warning screen even if the Failsafe Mode option is selected as "Not Set" and the Radio is restarted?

**Expected behavior**
Failsafe Not Set warning screen should not trigger on a model that has no Radio Failsafe settings.
But should trigger on a model with Failsafe Mode "Not Set" and Radio is restarted.

**General information**
 - OpenTX version: 2.3.0RC5
 - Hardware: Taranis X9D+

3djc, 2019-09-25:
You found a nice and technicaly chalenging issue ! Reproduced, working on a fix

MRC3742, 2019-09-25:
It was also challenging to describe but you were able to reproduce it, so I must have explained it correctly. I have a less critical one but will wait until this gets resolved before opening it.
Seems you are all quite busy getting this version ready for a full Release.

Thanks for your quick response to this issue. - Rich

MRC3742, 2019-09-28:
Closing as this problem has been resolved in #6829 and has been merged into Final Release 2.3.0

PR #6829 (2019-09-27): 3djc/fix 6826
Introduce delayed failsafe update for multi, as the module needs to be running for a proper check to happen. Previously, the module was not ready and the check simply bypassed

This fixes #6826
closed 2019-09-28 by MRC3742