opentx/opentx #6833: "Throttle trim idle only" effects elevator using cross trim when all trims at neutral

sande005, 2019-09-26:
With cross trims enabled, and all trims at neutral, checking "Throttle Trim Idle Only" introduces significant change to Elevator neutral output.  See attached simple model file illustrating.    Uncheck "Throttle trim idle only" to restore normal operation.  (Rename file to ".otx" for Companion).

**Expected behavior**
As with other cases involving this function, it should be aware of which trim is being used for throttle trim, and should not effect neutrals when trim is at "0".

**General information**
 - OpenTX version 2.2.4
 - Hardware: Frsky X9D+
 - OpenTX Companion 2.2.4
[crosstrim.txt](https://github.com/opentx/opentx/files/3658889/crosstrim.txt)

MRC3742, 2019-09-26:
Isn't this the same problem as discussed in #5635 ?
Looks like it was fixed in # 5664 but was reverted and Issue 5635 was reopened for discussion.
Also shows milestone has been changed from 2.3.0 to 2.3.x 16 days ago.

So i think this is a known problem?

kilrah, 2019-09-26:
Duplicate

closed 2019-09-26 by kilrah