opentx/opentx #6838: ACCST D16 not binding with integrated receiver Taranis X9D+ 2019

warbird11, 2019-09-27:
If I want to bind one of my mini quad with crazybee F3/F4 board and integrated FrSky receiver it does not bind. I’m using ACCST D16 mode in the new Taranis plus 2019.
If I use my irangex irx4 multi module and ACCST D16 it binds and also with my old X9D (no plus) it binds using ACCST D16.

An original FrSky X8R receiver can also be bind to the new X9D+ 2019. 

To reproduce just try to bind one of the mentioned boards (maybe all boards with integrated receiver) with the new X9D+ 2019 and ACCST D16

- OpenTX version: OpenTX 2.3.0RC5
- Hardware: X9D+ 2019
- OpenTX Companion 2.3.0 RC5

3djc, 2019-09-27:
We understand what you say, but this as nothing to do with OpenTX, but is rather a choice made by FrSky. Please contact them directly about that

closed 2019-09-27 by 3djc

warbird11, 2019-09-27:
You're fast! Wow ;-)
ok thx for the hint! Will do so....

j1and1, 2019-11-07:
@warbird11 Did you manage to fix this issue?

warbird11, 2019-11-07:
No! I soldered an external XM+ receiver to the board as a workaround. You can also use a multi protocol module but this was too uncomfortable for me.

j1and1, 2019-11-07:
Ok... @warbird11  Thank you for your reply! The new Taranis appears to be really picky on the RX side of things... as some of the generic china mini RX also are not binding to it.....