opentx/opentx #6839: cosmetical: duplications in list of contributors to opentx

gretel, 2019-09-27:
it's just a cosmetical issue.. the list of contributors in https://github.com/opentx/opentx/blob/2.3/CREDITS.txt has many dupes.
please have this sorted and uniquely permuted list:

```
2 Brains Design
A Harris
Aaron Stringer
Adam Brackney
Adam Chevalier
Adam Kalus
Adil Busaidy
Adilson Oliveira
Adolfo Perogio
Adolphe Montefiore
Adrian Andrei
Adrian De Zee
Adriano Del Ben
Adrien Gravouille
AIR-RC
Alain Labonne
Alain Pinard
Alan J Wasserman
Alan Thoren
Alastair Smith
Albert Blondin
Albert Herzog
Alberto Scotti
Albin Esbjörnsson
Aldo Midali
Alessandro Apostoli (Yaapu!)
Alessandro Mari
Alessandro Villa
Alex Calderhead
Alexander Mathis
Alexander Orlov
Alexander Pils
Alexander Röhlich
Alexander Wimmer
Alexandre d'Alton
Alexandre Rault
Alexis Pons
Alfredo Mac Donald
Algirdas Tamosiunas
Ali Ersan Gur
Alistair Sutherland
Allan Abbott
Aloft Hobbies
Alonzo Fretwell
Alvin Burke
Anatoly Bakalenko
Andrea Montefusco
Andreas Weitl
Andreja Rusjakovski
Andrew Duncan
Andrew Fazio
Andrew Fernie
Andrew Gardiner
Andrew Garello
Andrew Newton
Andrew Scerbina
Andrzej Weydmann
André Curci
André Lasseret
ANF King
Angel de la Dehesa Calles
Anindya Mukerjee
Anselm Schopplick
Anthony Burbidge
Anthony Hrabusicky
Anthony Rizzo
[... 1139 lines cut ...]
William Mckinstry
William Simon
William Tortorello
William Waters
Willie Swanepoel
Willyp
Wolfgang Kuhl
Wolfgang Lang
Wolfgang Rauser
Wolfgang Renisch
Wolfgang Schneider-Reif
Xavier Delepine
Xavier Mestelan Pinon
Yak
Yashiro Masayuki
Yeo Jonathan
Yves Berger
Zdenek Trojanek
ZEN Computer Services
Zhang Tianyu
Åke Brodin
Марков Алексей
Шарлаимов Константин
单 雷平
```

mpjf01, 2019-09-27:
Why? It's just added work and if someone donates many times and his name appears many times it's useful information.

gretel, 2019-09-28:
@mpjf01 in a plain text file - how can you tell who did donate multiple times and who is listed multiple times for a single donation?

kilrah, 2019-09-28:
`sort CREDITS.txt | uniq -c | sort -r` ?

closed 2020-03-27 by kilrah