opentx/opentx #6840: Menu navigation

RookFPV, 2019-09-27:
Bf.lua - When in the menu e.g. VTX I want to change e.g. channel, I can't slide down, I can only navigate only up

Crossfire.lua - Navigation up or down is reversed ... when I move the cursor up the menu goes down

ps. Why doesn't bf.lua display me in the new menu?

**General information**
Betaflight Script 1.3.0
 - OpenTX version (2.3.0)
 - Hardware Frsky X9D+
 - OpenTX Companion 2.3.0 windows

3djc, 2019-10-03:
Yeah, for some reason the script expect a + event to increment value, but a - to go to next field, which got me confused. Will fix the script tomorrow

kilrah, 2019-10-03:
That's becasue on X9D it is intuitive to press the DOWN button to go DOWN in the fields, but the DOWN button is -.

3djc, 2019-10-04:
Then we need to add another virtual event to distinguish inc/dec and previous/next

closed 2019-10-04 by kilrah