opentx/opentx #6841: Synchronize SD hangs

twistedwings, 2019-09-28:
Synchronize SD syncs forever...

[image]

After 1 hour I aborted the sync.

OpenTX 2.3.0
Taranis Q X7
Companion 2.3.0

KingKnecht, 2019-10-28:
Same here:
Version 2.3.1 Oct 5 2019
X10S Express

KingKnecht, 2019-10-28:
If my "SD" folder is empty on PC side, it works once. Next sync never finishes.

KingKnecht, 2019-10-29:
I have installed Companion on another PC (Win10Bit as well), same problem. From radio to PC is ok. From PC to radio is not working.

KingKnecht, 2019-10-29:
The debug log looks like this (includes manual termination because it hangs):
```
[D] findMassstoragePath():437 - Searching for "G://SOUNDS"
[W] FileSyncDialog::reject():71 - Terminating thread while Sync process was still running.
[W] unknown():0 - QThread: Destroyed while thread is still running
```

KingKnecht, 2019-10-30:
I have compiled and debugged v2.3.2 to check where the sync problem comes from.
In file `process_sync.cpp`
in function 
`QFileInfoList SyncProcess::dirInfoList(const QString & directory)`
the line 
`return QDir(directory).entryInfoList(m_dirIteratorFilters, flt, QDir::Name | QDir::DirsLast);`
never returns.

I made a video of the debug session:
[https://youtu.be/WLqDapUMDyU](https://youtu.be/WLqDapUMDyU)

Please, let me know if you need additional information.

KingKnecht, 2019-11-07:
Ok, I'm able to sync in both directions now:
[image]

I have removed the `QDir::Readable` flag. I don't know why this solves the problem neither why it is needed in the original code.

```
void reset()
        {
          excludeFilter = QStringLiteral(".*,System Volume*");
          includeFilter.clear();
          direction = SYNC_A2B_B2A;
          compareType = OVERWR_NEWER_IF_DIFF;
          maxFileSize = 2 * 1024 * 1024;
          logLevel = QtWarningMsg;
          flags = OPT_RECURSIVE;
          //dirFilterFlags = QDir::Dirs | QDir::Files | QDir::NoDotAndDotDot | QDir::Readable;
          dirFilterFlags = QDir::Dirs | QDir::Files | QDir::NoDotAndDotDot;
        }
```

kreidler, 2020-01-02:
Still same here:
Version 2.3.4
X9D+ / X7
Win10 Pro 64Bit
Bitdefender in use
But no chance to compile a version by myself.

Sync works once into an empty local directory. Does not matter where the folder is located.

PR #7258 (2020-01-03): Fix to some sdcard sync issues ?
This is a user proposed fix to https://github.com/opentx/opentx/issues/6841

Fixes #6841 ?
closed 2020-01-03 by bsongis