opentx/opentx #6842: RTC Battery Voltage wrong value

0815Creeper, 2019-09-28:
**Describe the bug**
on my x9e, the rtc voltage in the hardware tab is displayed to be 1.99V. When I check with a multimeter (with radio on), the true voltage reads as 3.0V. I got no problems with the date and time keeping. On every boot, the RTC-warning is displayed (which is massivley annoying)

**General information**
 - OpenTX version (e.g. 2.3.0)
 - Hardware (Frsky X9e)

3djc, 2019-09-28:
Working fine here with production X9E and 2.3.0, so different versions of X9E must exist

robustini, 2019-09-29:
Same here on Horus X12S, show 1.49V.

dev-fred, 2019-09-29:
Same here on X9D, shows 1.62V.
The error message RTC Battery low appeared just after a firmware update 2.3.0 (official release) + a message indicating  rewrite of the EEPROM after the restart of Opentx.

3djc, 2019-09-29:
Please indicate your real voltmeter measured  RTC bat voltage

dev-fred, 2019-09-29:
I have measured 2.625V on the  CR1220 cell.

3djc, 2019-09-30:
All this doesn't make much sense yet, as it would suggest that different batches uses different vref, which is news to us and does not make a lot of sense. So in the meantime, while we try to gather more data, a choice to disable the alarm will be added to 2.3.1 in radio/hardware screen

0815Creeper, 2019-09-30:
wouldnt it be better to add a way to calibrate the rtc voltage just as the battery voltage? or is that not possilbe?

3djc, 2019-09-30:
Not possible within current data model

dev-fred, 2019-09-30:
Thanks for this workaround, it's better than coming back in 2.2.4

dev-fred, 2019-09-30:
I have checked this new firmware 3.0.1on my X9D,
I have a new ligne Check RTC voltage with a selection box but I can't select this box to uncheck the default choice ?
I am able to uncheck Filtre ADC or ON/OFF Port série just after or Calib. bat above.

dev-fred, 2019-09-30:
Same behaviour in the simulator so nothing specific to my hardware

kilrah, 2019-10-01:
Will be OK in the next nightly, was an unrelated issue on that screen for the old X9D that got fixed in #6860 merged a couple of hours ago.

dev-fred, 2019-10-01:
Thanks a lot

OlivierJB, 2020-01-28:
I just upgraded 2 Taranis to latest (one +, one non +) fron older versions and both now have the RTC low warning. Any news on this issue?

kilrah, 2020-01-28:
Well, replace the battery?

OlivierJB, 2020-01-28:
Sure. But two transmitters that worked perfectly fine just before upgrades (bought a couple years apart) , somehow got failing batteries within 1/2 an hour or so after upgrade? This seems extremely unlikely.

I guess one possible explanations would be that perhaps the batteries were already failing before but the check was different,  or there was no check?  Anything changed between say 2.2 and 2.3.5 in that area?

Thanks.

3djc, 2020-01-28:
There was no check before

enriqueigl, 2020-01-29:
Hi i Have changed rtc battery today CR1220 3v and now i have rtc battery low in my x9e 2016.

3djc, 2020-01-29:
X9e has an issue, several versions seem to exist, but we cannot recognise those yet. So you have the option to disable it

enriqueigl, 2020-01-29:
Hi 3Djc. Where to disable? Thanks

3djc, 2020-01-29:
Radio/hardware

enriqueigl, 2020-01-29:
Ok, I go to see it!

enriqueigl, 2020-01-29:
ok. check rtc voltage disabled. RTC bat say 1.95v!!!!

3djc, 2020-01-29:
Known issues on some x9e, as said earlier

enriqueigl, 2020-01-29:
no previsions to change it?

kilrah, 2020-01-30:
We cannot know what version has what.

0815Creeper, 2020-01-30:
Well, how about just letting the user calibrate it just as the main battery?

3djc, 2020-01-30:
Can't be done before 2.4, and would not be a great solution either. We are still trying to find if we can recognize those various versions

enriqueigl, 2020-01-30:
The idea of 0815Creeper is good. So do not play too much code is already open to the user, and at the same time allows you to control that battery. After all the work you do, this doesn't matter since it can be deactivated, as I have done.

Rob2160, 2020-02-02:
Posting this in case it can help diagnose the issue. 

Recently updated an X9D+ and an X9E to 2.3.5 from 2.2.4.

The X9D+ immediately showed the RTC low battery warning. (1.3 volts indicating)
The X9E was fine - showing 2.3 volts. 

I bought two new batteries CR 1220 and fitted them to each radio. 

The X9D+ was fine - now showing over 3 Volts and no warning. 
The X9E gave the RTC Low battery warning and with the new battery was showing 1.8 volts.  

I swapped the batteries over between radios to check and the X9D+ was still good with the second battery - over 3 volts. 

The second new battery (confirmed as good in the X9D) now in the X9E still gave the low voltage warning and 1.8 volts. 

Both new batteries tested to 3.2 volts on a multimeter

So the X9E clearly has an issue.  

Now here is the interesting part.  When I fitted the original X9E battery (showing 2.8 volts on a multimeter) the warning disappears and it shows 2.3 volts in the radio.

Perhaps there is an issue if the voltage is over 3 but ok if under this?   Just guessing based on my results.  

My X9E build is AU US AH  Date 30/12/2015 and has worked flawlessly for years.

3djc, 2020-02-02:
Not ok under 3v, since 2.8v translated to 2.3v

Rob2160, 2020-02-02:
True,  I guess that is a calibration issue though but that is not consistent as 3.2 shows as 1.8 which makes no sense at all. 

Perhaps I worded it wrong.   There may be different issues above and below a certain threshold.  Just a guess though.

PR #7382 (2020-02-18): Fix X9E RTC batt voltage readout
This fixes #6842
closed 2020-02-18 by bsongis