opentx/opentx #6843: Companion sync misses the log level  "Updated/Created"

notlebowski, 2019-09-28:
The Companion sync in 2.3.0 has different logging capabilities:

- Skipped
- Created
- Updated
- Errors only

When I sync the SD-Card with my computer I would like to see the files it adds **and** the ones it updates. When I set it to "updated" I will never see the files it created and the other way around.

Could the log level "Created/Updated" be added as a log level?

notlebowski, 2020-03-31:
Please let me know if the problem is understood.
If not, I will try to eloborate on the issue.

haerel, 2020-06-21:
v.2.3.9
I think how notlebowski.
In my opinion, it is of little use if only one log level can be seen in the detail window (Created, Updated or Errors). All log levels (except Skipped) should be displayed when synchronizing. At some point it was like that.

notlebowski, 2020-06-22:
Indeed. Customizing the logging isn't even necessary as that would be a one size fits all.
It's however impossible to set it that way.

notlebowski, 2020-12-19:
it would be nice if the log contained both updated as created files.
In fact it would be even good if it would be the only way it would log.

closed 2021-09-22 by stale[bot]