opentx/opentx #6847: Companion on 32 bit linux

nemoxeba, 2019-09-28:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

nemoxeba, 2019-09-28:
will it still support 32 bit linux?
thanks

kilrah, 2019-09-28:
You can build from source, but we're not supplying 32-bit builds anymore.

closed 2019-09-28 by kilrah