opentx/opentx #6850: LUA interpreter for the new X9D+ 2019

m-rik-prog, 2019-09-29:
Good morning, sir,
My Luac application works on Horus 10/12 and Q X7 but did not work on X9D+ due to insufficient memory (used for larger screen pixels than Q X7).
Now users tell me that on the new Taranis X9D Plus 2019 (new processor more powerful and with more ROM and RAM) the application finally loads without more problems of insufficient memory (just like with the Firmware Simulator that has no memory problems using the one of the PC on which it runs) and it is possible to navigate back and forth in the different pages.
But the telemetry data are not loaded and displayed even after the update to the final release 2.3

Does anyone know if it's a known bug of the LUA interpreter of the new X9D Plus 2019 or should we introduce some changes in the LUA application for the telemetry data management (sportTelemetryPush / sportTelemetryPop) or for the new roller that replaced the buttons of the old X9D transmitters can be the cause of no sport telemetric communication?

Note: If I have to add specific code for the new X9D Plus 2019 my application will no longer work on Q X7 because I have reached the maximum limit to work in the little RAM of Q X7.

kilrah, 2019-09-29:
Are these people using ACCESS or ACCST?  Telemetry works differently with ACCESS given that multiple receivers can be telemetry sources/destination, so you'll likely have to rewrite some things specifically for ACCESS systems.

> Note: If I have to add specific code for the new X9D Plus 2019 my application will no longer work on Q X7 because I have reached the maximum limit to work in the little RAM of Q X7.

You can always distribute separate versions for different radios.

m-rik-prog, 2019-09-29:
Thanks for the information!
Where can I find the documentation that explains what are the differences in the data to be sent with sportTelemetryPush and in the data received with sportTelemetryPop with ACCESS compared to the same data that I have transmitted and received so far with ACCST in order to implement the new ACCESS version?
From what I understand it is only a byte (or some bit) added in addition that identifies the receiver. But where and how should this byte be added?

kilrah, 2019-09-29:
There is no doc as of yet.

m-rik-prog, 2019-09-29:
Thank you.
From what I understand I can not give users who are complaining, a date when the problem could be solved.

kilrah, 2019-09-29:
Well, even we don't really know all about it. So you'll likely want to source an ACCESS radio and set of equipment and start to investigate.

bsongis, 2019-10-09:
I have had access to your script. I hope it's already released (or you will release) it as GPL as I am the original author of what you copied.

```lua
local function telemetryWrite(field, value)
  return sportTelemetryPush(0x1B, 0x31, field, 0x01 + value*256)
end
```

The dataID (3rd parameter) is searched in the list of the sensors inside the Model / Telemetry page. 
Would you confirm the sensor with this dataID is present in this page?

bsongis, 2019-11-07:
I have seen no other reports on that topic, so I guess the issue is in your script. I can propose to help if you accept to send me one of your sensors for a couple of weeks. That's the maximum I can do.

bsongis, 2019-11-12:
I received your sensor.

1) I plugged it on a G-RX8R receiver running the latest ACCESS firmware. The receiver is bound to a X9D+ 2019 radio
- I discovered the sensors => OK
- I executed the lua script => I saw all parameters and was able to modify some of them at random

2) Then I connected it to a X6R ACCST receiver, bound to the same X9D+ 2019 radio
- I discovered the sensors => OK
- I executed the lua script => I saw all parameters and was able to modify some of them at random

On my X9D+ 2019 I am running the latest nightly build (which will be the next 2.3.2).

bsongis, 2019-12-23:
I received a message, saying that the OP succeeded to make it work on first attempt (no issue). I would like to thank all people who spent their time trying to find a solution.

closed 2019-12-23 by bsongis