opentx/opentx #6851: ACT switch setting not recognized in companion

Tygrys-1, 2019-09-29:
When ACT is set as a switch option in the simmulator or on the radio (i.e. on the mixer), the companion editor changes it to --- after loading the settings. Also, there is no ACT setting in the companion editor.

**To Reproduce**
Set ACT as an switch option in the radio or simmulator, save the settings and open them in the companion.

**Expected behavior**
Companion should have the ACT setting added to the switch options and stay consistent with the radio. An explanation to this option should be added to the documentation.

**General information**
 - OpenTX version (2.3.0)
 - Hardware (Frsky X9 Lite Pro)
 - OpenTX Companion (2.3.0 on Win64)

3djc, 2019-09-29:
What exactly do you call ACT ? Not sure we understand what you mean

Tygrys-1, 2019-09-29:
It is the one of the switch options to choose, like SA-, ON or TELE. I think it is new to 2.3.
For what I gather it is acting like momentary pulse of "true", when the rule or the condition is activated.
Just like ON is true all the tile, ACT is true for a moment.
At least I have this setting in the radio ;-)

Tygrys-1, 2019-09-29:
[image]

notlebowski, 2019-09-29:
@Kilrah
What does Act do?

3djc, 2019-09-29:
Any action on the radio for a short period of time

notlebowski, 2019-09-29:
Hey.   I requested that years ago

notlebowski, 2019-09-29:
I just looked at that request and now I remember...
I wanted to use it to enable a throttle safety, but I can't do it the way it is implemented now because "delay" can't be more than 25 seconds...

Otherwise I could use this trigger to invoke throttle safety

AND  !Act delay(180s)

At the time I was looking for a trigger that would become true if the inactivity becomes true.
I may still be able to use this trigger for that using the timer function...

BTW...    Why is delay limited to 25s and is it hard to make that longer?

kilrah, 2019-09-29:
One 8-byte value available with 0.1s increments.

notlebowski, 2019-09-30:
@kilrah 
I'm considering to use timer3 and set that to 3 minutes and let ACT reset that each time.
Whenever timer3 = 0 I can use that as a logical switch and that could invoke throttle safety

I think it's a bit ugly to use a timer for that. Do you have a tip for a more elegant solution to get a non-activity trigger that acts like:

L6 ( Thr < -95 ) AND !Act↓ Delay(180s)

It's an extra safety measure that will invoke a throttle safety after 3 minutes of inactivity if you forget the radio and want to avoid that someone bumps the throttle stick.

kilrah, 2019-09-30:
You can stack 2 LSs and declare that 50sec is enough...

notlebowski, 2019-09-30:
@kilrah Yes, I think I will do that. 50s is a long time to have both inactivity and no throttle.

commit 4f614850 (2019-09-30): Add "Act" switch to companion (fix #6851) (#6854)
closed 2019-09-30 by kilrah