opentx/opentx #6852: changed value with lua script has no effect

hsuq510, 2019-09-29:
- OpenTX version 2.3 GA
 - Hardware Frsky X9D+

Changing a value with the minus button on a x9d+ has no effect, the number switch back to the previous value after release the minus button.

3djc, 2019-09-29:
What script is that ?

RookFPV, 2019-09-29:
> What script is that ?

This problem occurs in the betaflight script,
you have to hold down the button for a long time to return to the value down

hsuq510, 2019-09-29:
is betaflight script lua :
https://github.com/betaflight/betaflight-tx-lua-scripts/blob/master/src/SCRIPTS/BF/events.lua

RookFPV, 2019-10-03:
Will this problem be resolved in 2.3.1?

3djc, 2019-10-03:
We seem to have established in #6814 that the events are triggered correctly, so unless somebody diagnoses exactly what is happening in this script, it probably won’t be fixed

RipperDrone, 2019-10-03:
Issue confirmed on X9DP 2013, OTX 2.3.1N5, BF telemetry script that comes with BF4.1 or: https://github.com/betaflight/betaflight-tx-lua-scripts/releases

To replicate:
1. Flash BF 4.1 on quad
2. Flash OTX 2.3.1N5 on RC, including latest BF lua
3. Long press <Page> to enter telemetry screen
4. Press '+' button to enter PID telemetry page
5. Navigate with +/- and edit any value by pressing <Enter>
6. Try to decrease this value by short pressing '-'

--> Value is decreased, then displayed for a fraction of a second, then increased back to what it was. There seems to be a small bug in evaluating the button press/release events correctly :-)

3djc, 2019-10-03:
Then I suggest you try the event tester as described in #6814

RipperDrone, 2019-10-03:
@3djc Happy to do so - have downloaded both tester lua files and tried to execute on Taranis --> error 7: unexpected symbol near '>'

Am I doing it wrong?

RookFPV, 2019-10-03:
By the way. In the crossfire script, menu navigation works the other way around when you press "+" in the menu goes to "-"

RipperDrone, 2019-10-03:
@RookFPV True - but different topic. You might want to open a separate Issue report?

RookFPV, 2019-10-03:
I already opened it before but I think it was omitted, #6840

RipperDrone, 2019-10-03:
@3djc Have managed to download RAW version of lua and it executed on my X9DP 2013. +/- events are shown properly. +/- is also working ok in all telemetry screens, as long as you navigate only. Just when you edit any value, then '-' fails by somehow issuing a '-' first, followed by an uncommanded '+'. Pls see yourself by plaing through the sequence I described before... strange...

3djc, 2019-10-03:
I cannot since I do not have a x9d(+), but clearly events seem to be ok. someone needs to investigate what is going on in the script

RipperDrone, 2019-10-03:
To update everybody here: 

Issue has to do with land of confusion caused by the plethora of supported RCs with buttons / wheels / knobs etc. OTX 2.3 has made changes to event handler which render BF Lua partly incompatible for some RCs.

OpenTX has introduced virtual events which BF lua script needs to adopt now as an abstraction layer. @raphaelcoeffic is the boss in this game on BF side and supposedly will take care of it... :-)

RipperDrone, 2019-10-03:
for an interim workaround - see referenced issue #245 above.

raphaelcoeffic, 2019-10-09:
@RipperDrone I just added a fix in the BF scripts and improved the longer term solution. Please test these 2 PRs and give feedback there!

3djc, 2019-10-09:
Closing as I have been informed by  BF team that this is resolved on their side (thx BF guys ;))

closed 2019-10-09 by 3djc

RipperDrone, 2019-10-09:
@raphaelcoeffic happy to test. LAst JEnkins files are dated Oct 2nd so this does not et contain your last PRs, right? Can I just download the .zip from the GitHub BF_tx_lua repository and copy it on my Taranis to test? Or does it need any compilation or such? Have been using releases and Jenkins builds only so far...

Seems @mikeller has tested the 2 PRs and confirmed on X9D+ anyway in the meantime, then pbly you don't need more tests on X9D+?

raphaelcoeffic, 2019-10-09:
@RipperDrone just copy everything in `src/` into your SD card. No processing needed here. And the more tests we have, the better it is.