opentx/opentx #6853: Curve points in one curve changing when editing another curve

sloper-lloyd, 2019-09-29:
2.2.2 and 2.2.4 same issue
X10S
Opentx Companion

When I edit a curve it corrupts the other curves.
First discovered this when the battery died whilst I was programming my tx when I recharged and restarted it the curve points had changed. So I tried to update firmware to 2.2.4

Now curves points are changing when I edit another curve. I can corrupt 1 or many other curves.

Any ideas I see this as critical, what should I do.

kilrah, 2019-09-29:
You say this is in companion, but you talk of battery... so where is it?

sloper-lloyd, 2019-09-29:
No it doesn't happen using companion. I program using the tx usually.
I have spent hours on this trying to fix it. I got it to work eventually. But there is definitely a bug where data points seem to 'push' onto other curves and get changed after a power cycle.

sloper-lloyd, 2019-09-29:
Noone else seen this its a pretty major issue?
I use up to 32 curves on some of my multi surface winged gliders. Maybe others don't use so many curves with up to 17 data points

kilrah, 2019-09-29:
We have never heard of that before.

Please zip and send the RADIO and MODELS folders from the radio's SD card, and give us some instructions to reproduce.

sloper-lloyd, 2019-09-29:
Think the issue occurs if you have a curve set with say 11-13 points then increase the number of points to a higher number say 15-17. If the curve is between other curves (and points are adjusted) then the adjacent curves are messed up. That's the only way I can describe how it occured.

kilrah, 2019-09-30:
Can we have your files as mentioned in my previous posts?

sloper-lloyd, 2019-09-30:
The model in question is Freestyler 5

kilrah, 2019-09-30:
Can't send mail attachments to github.

sloper-lloyd, 2019-09-30:
Where do I send the attachment ?

sloper-lloyd, 2019-09-30:
[29-09-19_models_settings.zip](https://github.com/opentx/opentx/files/3672600/29-09-19_models_settings.zip)

sloper-lloyd, 2019-10-01:
Programming using tx not Companion on X10S. Try playing with the curves, add some numbers, increase the number of data points to 17 and power cycle tx. The data point values change and other curves are affected also

closed 2020-05-30 by bsongis

BeckyYork, 2020-07-14:
This is still a problem on Radiomaster TX16S - while editing Curves on the radio.
Editing CV7 & CV8 changes the curves on CV9, CV10, CV11

3djc, 2020-07-14:
Video possible?

BeckyYork, 2020-07-14:
I just tried to replicate it after fixing CV9,10,11 but couldn't. seems to be intermittent and I don't seem to be able to break it again. I'll keep an eye on it.

PR #7643 (2020-05-30): Fix  6853
Fix #6853. This issue is affecting radios with rotary encoders

When increasing the number of points of the curve, it has to be done in increments of one. If the rotary goes faster then the new points are also copied to adjacent curves. So lets force low rotary speed when changing the number points of the curves