opentx/opentx #6860: From Mike: Shift in X9D+ sources

kilrah, 2019-09-29:
Telemetry Display Values X9D Plus - 2.3.0.
“Time” displays TX battery value
“GPS” displays TX time. Not sure where "GPS" is coming from because there is no "GPS" sensor showing on the Telemetry page. 
Not sure if there are other examples.

RipperDrone, 2019-09-30:
confirmed today on X9DP 2013, GPS displays time.

closed 2019-10-01 by kilrah

RipperDrone, 2019-10-01:
@kilrah Still not ok, tested with 2.3.1 N2 Nightly and matching firmware. GPS still displaying time...

Found that there were Travis errors in the PR. How's the procedure: PRs are merged even though Travis checks fail? Would just like to understand theprocedure, so I start testing at the correct point in time... :-)

kilrah, 2019-10-01:
N2 was made before this was merged.

RipperDrone, 2019-10-01:
Good point, I haven't checked time details. Still curious: Commits can be merged ok even though Travis checks are failing? So whenever build date of a Nightly is later than Pull Request X, then it will be included?

kilrah, 2019-10-01:
If you look in #6857 you will see "2 checks passed", which means the changes introduced by that PR would build correctly. 
It just so happens that something else had broken the commit tests on the 2.3.1 branch in the meantime, so anything done on that branch after it would fail automatically until it was fixed in https://github.com/opentx/opentx/commit/8b598be7bfc7b6999315d823f5d4d1eddf81b4b1.

RipperDrone, 2019-10-03:
looks good on X9DP now. Can close?

commit e74f14cc (2019-10-01): Fix shift in sources on X9DP pre-2019 and X-Lite and hardware page (fix #6860 fix#6856)

* Fix shift in sources on X9DP pre-2019

* Probably better fix, but needs to be carried to Xlite if approved

* Fix Xlite strings

* Forgot to update conversions too