opentx/opentx #6862: Companion telemetry simulator not usable with "real" sensors on 2.3

kilrah, 2019-09-30:
Lacks support for keeping the source intact and feeding the same one

offer-shmuely, 2019-10-09:
can you tag it with “companion” label?

George-111, 2019-11-08:
I'm getting the same issue with Companion 2.3.1, simulating telemetry of my QX7S under Windows 10 1903. 
None of the manually entered telemetry values for RSSI, RxBt, Cels, Alt etc. are posted to the simulated radio or logical switches with checked "Simulate".

T3chDad, 2020-01-06:
Telemetry simulator not working in 2.3.4.  Multiple, radio profiles, multiple models.  Windows 10 1903 or  Insider 15936.

kilrah, 2020-01-06:
Yeah that's the whole point of this issue.

offer-shmuely, 2020-01-06:
Is there estimation for a fix?

T3chDad, 2020-01-06:
> Yeah that's the whole point of this ticket.

My apologies, I was just verifying this issue still exists in the stable version released in mid December.  It pretty much makes LUA debugging impossible.  Thank you for your efforts.

guillaumevaret, 2020-01-27:
Hi everybody. I will add some elements for the diagnostic of the problem.
 - post #7328 : french duplicate of this one with more infos, telemetry problems :  it seems that it's RSSI 248 that is wrong => should be RSSI 25 to makes works all the others sensors.
  - post #7327, telemetry transfers through companion and directly go back through the radio. The edition of each sensor, changing the type and put it back fix the problem 1
 
the instance of the ID can take the (0 to100) values when just discovered or in companion, but goes from 1 to 32 in opentx. 

from the 2.3.6 companion, different vues of the 2 sames sensors in the
- telemetry tab on companion
- telemetry simulator on companion
- simulator of opentx edition on companion
- simulator of telemetry tab on companion

[image]

so it seems that the problem come from the conversion of the ID number + instance  (telemetry tab on companion and principal telemetry screen on opentx) into a unique ID number (simulator on companion and edition of sensor on opentx)
-  removing sometimes 1 to the unique ID number.
-  taking the instance for unique ID  number instead of ID number

gastolectric, 2020-01-30:
For anyone else following this 'issue' - I have found that amending the reference value in the telemetry simulator to match the "Instance" ID's on the telemetry menu page resolves the issue of telemetry not simulating in Companion 2.3.x for Horus radios. 

[image]

LapinFou, 2020-01-30:
If this can be any helpful, it seems there is an offset of 223 on all instance. After lunching the simulator, if you do the subtraction of -223, the sensor will work.
It has been found by a French guy here: [#107929](http://frskytaranis.forumactif.org/t9810-probleme-simulateur-telemetrie-companion-2-3-4#107929)

guillaumevaret, 2020-01-30:
On the side of the horus, each time I transfer my model to the radio,  I have a problem with all the sensors in ID 1 not recognized anymore (so ID224 coming from the PX4), I edit the sensor and double enter on the "type", it fix the problem and put back the ID on the telemetry screen to 0 (=> ID = 1 on the ID edit).

sam2b, 2020-02-10:
@LapinFou thanks for attention on this! At least there is a workaround for now. :) 👍

offer-shmuely, 2020-04-10:
any estimation for a fix?
we currently need to debug widgets on companion 2.2, 
however the X10 itself is on an other version (2.3.7)

3djc, 2020-04-10:
Should be ok in 2.3.7

offer-shmuely, 2020-04-11:
I found the cell voltage are working, but the RSSI does not

offer-shmuely, 2020-10-10:
this issue not need to be stale
it working very good, no issues anymore

can be closed

closed 2020-10-10 by 3djc