opentx/opentx #6866: 2.3.0 RAS Issues

MIKE-DAILEY, 2019-09-30:
There might be something not right with 2.3.0 RAS:

RAS display ---/---on systems using Hardware Menu
It is 0/--- on the X9D Plus 2019 
It is 0/--- on the X-Lite - Pro RAS working with external antenna selected and RAS script works. 
It is ---/0 on the X9D Plus – RAS number works, but not RAS script.  Should the RAS number be on the right or left?  

RAS on Statistics /Analogs 
X12S Developer – RAS not working stays at 0.

robustini, 2019-10-01:
Already reported...

https://github.com/opentx/opentx/issues/6775

kilrah, 2019-10-01:
Indeed, duplicate

closed 2019-10-01 by kilrah

MIKE-DAILEY, 2019-10-01:
That does not cover this part:

RAS display ---/---on systems using Hardware Menu
It is 0/--- on the X9D Plus 2019
It is 0/--- on the X-Lite - Pro RAS working with external antenna selected and RAS script works.
It is ---/0 on the X9D Plus – RAS number works, but not RAS script. Should the RAS number be on the right or left?

kilrah, 2019-10-01:
It's supposed to be Internal/External

MIKE-DAILEY, 2019-10-01:
All three systems were using the external antenna?

kilrah, 2019-10-01:
Internal/External **module**.

MIKE-DAILEY, 2019-10-01:
All three were on the internal module.  So the X9D Plus should be 0/--- not ---/0?

kilrah, 2019-10-01:
Yeah, as I said everything need reviewing.

VeedubDan, 2020-04-01:
Any update on this yet? My x9d+ is still showing - - - /0 pm the latest firmware. The 0  does work for the internal module though, if I shield antenna it goes up to 16.

MIKE-DAILEY, 2020-04-01:
Working correctly now in the latest 2.3.8 nightly.