opentx/opentx #6870: OTX 2.3 EVT_VIRTUAL_MENU not defined on Jumper T12?

teckel12, 2019-09-30:
Not sure if it's on purpose or an oversight, but `EVT_VIRTUAL_MENU` is not defined in Lua on the Jumper T12 in OpenTX 2.3.  It seems a [test script](https://github.com/opentx/opentx/blob/2.3/radio/util/lua/lua_event_test.lua) was created to verify all the basic functions are available on each target.

Disclosure, I don't have the Jumper T12, I'm testing this from the Companion simulator.

3djc, 2019-10-01:
True, is is the current situation:

EVT_VIRTUAL_MENU |   |  
EVT_VIRTUAL_MENU_LONG |   |  
  
EVT_VIRTUAL_NEXT | KEY_RIGHT |  
EVT_VIRTUAL_NEXT_REPT | KEY_RIGHT LONG
EVT_VIRTUAL_PREVIOUS | KEY_LEFT |  
EVT_VIRTUAL_PREVIOUS_REPT | KEY_LEFT LONG
  
EVT_EXIT_BREAK | KEY_EXIT |  
EVT_VIRTUAL_ENTER | KEY_ENTER |  
EVT_VIRTUAL_ENTER_LONG | KEY_ENTER LONG
 
EVT_VIRTUAL_PREVIOUS_PAGE | KEY_UP LONG
EVT_VIRTUAL_NEXT_PAGE | KEY_DOWN LONG

What we could do is:

EVT_VIRTUAL_MENU |   KEY_DOWN
EVT_VIRTUAL_MENU_LONG |   KEY_DOWN LONG
  
EVT_VIRTUAL_NEXT | KEY_RIGHT
EVT_VIRTUAL_NEXT_REPT | KEY_RIGHT LONG
EVT_VIRTUAL_PREVIOUS | KEY_LEFT 
EVT_VIRTUAL_PREVIOUS_REPT | KEY_LEFT LONG
  
EVT_EXIT_BREAK | KEY_EXIT
EVT_VIRTUAL_ENTER | KEY_ENTER 
EVT_VIRTUAL_ENTER_LONG | KEY_ENTER LONG
 
EVT_VIRTUAL_PREVIOUS_PAGE | KEY_UP 
EVT_VIRTUAL_NEXT_PAGE | KEY_UP LONG

teckel12, 2019-10-01:
@3djc That will work!

It would be ideal if every transmitter would expose these basic control key events to Lua. That would also include Horus widgets, but that's a different issue.

closed 2019-10-02 by 3djc