opentx/opentx #6873: 2.3.0 on Taranis  X7 Switches Si and SJ for p400 and p401. Switch changes are not detected

pelle17b, 2019-09-30:
**Describe the bug**
I have a Taranis X7 that is modified with 2 position toggle switches connected to p400 and p401. I have configured Si and SJ as 2POS in the HARDWARE menu.

1. No matter what position the switches are in I get a switch warning for Si and SJ when i start the radio.
2. If I configure any off the Si and SI switches as the SOURCE in the INPUTS menu for a model, there is no reaction when the configured switch is toggled. 
 
**To Reproduce**
As above

**Expected behavior**
To be able to detect changes for Si and SJ switches similar to other switches. 

**General information**
 - OpenTX version 2.3.0
 - Hardware Taranis X7

bsongis, 2019-10-01:
Would you please check that this issue is fixed in the next nightly build? Thanks!

pelle17b, 2019-10-01:
Will do. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

pelle17b, 2019-10-02:
Tested with Nightly 2.3 1N3.
Still both Si and SJ are configured as 2POS in the HARDWARE menu.

Si works as expected as far as I can see. 
- Switch changes are detected in INPUTS menu as expected, and it is working when used in LOGICAL SWITCHES menu.
- Si is present and working in HARDWARE - Debug - [KEYS] menu
- Switch warning is working as expected for Si when radio is started and models changed (tested both ways).

SJ does not work.
- It is not present in the HARDWARE - Debug - [KEYS] menu
- Switch changes for SJ are not detected in INPUTS menu or if it is used in LOGICAL SWITCHES menu.
- Switch warning is gone when radio is started (i.e. not as before), but it does not appear if SJ is flipped.

3djc, 2019-10-02:
Yes, sorry about that. SJ was looking for the wrong pin, fixed it, will be in next nightly

commit 7a0a6a93 (2019-10-01): Fixes #6873
closed 2019-10-02 by 3djc