opentx/opentx #6876: [2.3.0] - ISRM D16, missing antenna selection?

robustini, 2019-10-01:
I don't know if it's a bug, coming from the X12S where it was possible to select the antenna I don't see this option in the ISRM module set in D16 on Horus X10S Express.

**General information**
 - OpenTX: V2.3.0
 - Hardware: Horus X10S Express
 - OpenTX Companion. irrilevant

[image]

ParkerEde, 2019-10-01:
Hi, since 2.3.0 it's possible to set the antenna configuration in hardwaresettings . You can choose between "internal", "external" "ask" and "per model".
If you want to set it in the modelconfiguration, please choose "per model". X10 can only use internal or external antenna. Not both at the same time.

Regards

robustini, 2019-10-01:
I don't see this option inside "HARDWARE" on my X10S Express with 2.3.0.

kilrah, 2019-10-01:
For ACECSS modules the antenna gets selected in the "Options" page.

closed 2019-10-01 by kilrah

robustini, 2019-10-01:
The antennaa selection is available only by selecting ACCESS protocol, it does not seem very correct, if i use D16 should be visible already in the model menu.... no?

kilrah, 2019-10-01:
It's not realted to the protocol, it's related to the module type.

robustini, 2019-10-01:
When i've ISRM (the module) on the screen and D16 (the protocol) there's no option to select the antenna, this i mean.
"Option" appears only when the ACCESS protocol is set! :)

kilrah, 2019-10-01:
You have it in your capture above...

[image]

robustini, 2019-10-01:
Damn, you're absolutely right, I've seen it now, I'm sorry! :P

pandjarov, 2022-07-06:
When the checkbox is placed does the external antenna work with the two internals or they get switched off?

robustini, 2022-07-07:
> switched off

Switched off.

brainbubblersbest, 2022-07-07:
no i dont think so. are you shure?
ixjt worked with both antennas. (one in and one extern)
isrm has two intern and one extern.
isnt it that only one internal antenna switches off and the external antenna is used with the other internal antenna?

robustini, 2022-07-07:
ISRM does not make diversity between external and internal.

brainbubblersbest, 2022-07-07:
So whats the purpose of a single 2,4ghz dipole compared to
two 2,4ghz pcb antennas? Have you compared the rssi?

robustini, 2022-07-07:
> So whats the purpose of a single 2,4ghz dipole compared to two 2,4ghz pcb antennas? Have you compared the rssi/range?
> 
> I believe you but thats a bit weird, because the ixjt used both.

I haven't noticed many differences between internal and external.

Hlaalu11, 2023-06-09:
>  in the first case removing the external one could damage the module without any warning since currently the RAS is not working
> [image]

Forgive my ignorance, but how does removing the external antenna cause harm to the module, if RAS is not active?