opentx/opentx #6881: ACCESS: Support more than 3 receivers

robertlacroix, 2019-10-02:
Is there a technical limitation that prevents supporting more than 3 receivers with ACCESS? On ACCST I’ve been running all my 10+ quads on one model and it’s been working great. It seems like now I would have to create n/3 models and keep them in sync when I make changes. That’s a lot less convenient and a step backwards compared to ACCST imho.

3djc, 2019-10-02:
This question is really for FrSky, please contact them directly

closed 2019-10-02 by 3djc

kilrah, 2019-10-02:
You can bind and control as many receivers as you want with one model, the limit of 3 is for those that are active and streaming telemetry back to the radio at the same time.

robertlacroix, 2019-10-02:
@kilrah hat’s true, but you have to go into the bind menu and bind a different one every time you switch quads in the worst case. Currently on ACCST you bind all 10 a single time and then just plug in a battery and go, without having to do anything on the radio.

kilrah, 2019-10-02:
No, you can do the same here. It doesn't matter if the receiver is actually stored in the OpenTX menu or not (apart from when you need to set options on it).

robertlacroix, 2019-10-03:
@kilrah ah awesome, that’s good. Thanks!