opentx/opentx #6884: Where is the baudrate Setting in 2.3.0?

ps915, 2019-10-03:
I noticed, that the baudrate setting on the "hardware" page is gone!
Where can i find it? On which baudrates does 2.3.0 run now?

kilrah, 2019-10-03:
What radio, what context, what feature are you looking at,...

ps915, 2019-10-03:
I am looking at the "baudrate setting" of the Taranis QX7 which was on the "hardware" page. I want to lower the baudrate to 115200 to use the Crossfire TX Module without telemetry warnings.

3djc, 2019-10-03:
Then same place as before

closed 2019-10-03 by 3djc

ps915, 2019-10-03:
Ok, my bad! Thanks

Raziel1989, 2021-08-12:
Hello! In my opinion it's not in the same place before. It's missing and I can't find it ahywhere :( 
Open TX 2.3.15

3djc, 2021-08-12:
@Raziel1989 what radio? And what 2.3.15, you mean nightly?

Raziel1989, 2021-08-12:
@3djc Taranis QX7 Access. I'm trying to configure Crossfire module and I get the syntax error. In some other topic I found that changing tha max bauds will fix the syntax error problem.

Yes, Nightly build 2.3.15. Schould I try official one ?

3djc, 2021-08-12:
I'll have a look at that model tomorow

3djc, 2021-08-13:
@Raziel1989 X7 ACCESS does not display the option, and never has apparently. I'll add it, and you can check tomorows nightly (549) to see if it helps your issue

Raziel1989, 2021-08-13:
@3djc Meanwhile I fixed my problem so maybe no update is needed.
This article helped me: https://fpvfrenzy.com/tbs-crossfire-setup/ 
In the middle of article there is sentence:
"f that still doesn’t do the trick, download these scripts by GitHub user OhmEye and replace the existing scripts in the Crossfire folder with these. Run the script again and it should now work."
And it worked for me. Totally don't know what is the difference between this files and original ones but it works. No more syntax error showing while I'm trying to configure crossfire RX or TX. Today I'll do my first test flight so we wille se if there are any telemetry warnings.

alec-bip, 2021-12-29:
Hi everyone,
I have a problem on my Taranis QX7 when it comes to adding the TBS CROSSFIRE MICRO TX II. First of all I have soldered the little piece according to videos and forums I saw and then I should be able to change : external RF —> CRSF. However, this option does not appear when I turn the right wheel. Therefore, some forums and videos I saw told me to change “max bauds” from 400000 to 115200. However, the “max bauds” does not appear in the hardware page (6/7). I have the version 2.3.0. 
Please help me I am desperate

kilrah, 2021-12-30:
You likely have FrSky's factory firmware, that has CRSF taken out. Flash regular OpenTX.

kort0175, 2022-01-04:
I am having the same problem with my X9D plus. Flashed 2.3.14 and it says that version on the last tab but I have no max baud line in the hardware tab.

3djc, 2022-01-04:
> I am having the same problem with my X9D plus. Flashed 2.3.14 and it says that version on the last tab but I have no max baud line in the hardware tab.

Nor should you, as far as we know all x9d+ are capable of 400k, hence the reduced speed at 115k is not offered

kort0175, 2022-01-04:
The problem is when I try to use the Lua script for my ELRS module it hangs

3djc, 2022-01-04:
Well the issue was never reported for any crossfire module that use the same protocol, so it could be your module ?

kort0175, 2022-01-04:
Maybe. I will remove and check pins tonight.

3djc, 2022-01-04:
I think it was added to all radio in nightlies because of similar complains, so you migh want to try a nightly (they are currently extremely stable)

kort0175, 2022-01-04:
Will do. Thank you.

kort0175, 2022-01-04:
Ok, tuns out it wasnt any of those things. I pulled it out and flashed it by the USB to UART method (which is how it was pinned, go figure) and am now able to get the LUA script to run. I tested it on both 400k and 115k. Both worked. Now I just need to go back to a stable release. Thank you for your help. It got me pointed in the right direction!

poppel23, 2022-01-29:
> Ok, tuns out it wasnt any of those things. I pulled it out and flashed it by the USB to UART method (which is how it was pinned, go figure) and am now able to get the LUA script to run. I tested it [...]

Dear kort,
facing the same problem. running otx 2.3.14 on an x9d+ and trying to flash elrs 2.2 using the transmitter. As a result im getting a firmware update error "NoSync". The newest LUA is stuck in "Loading". Could you explain what u meant with flashing via "USB to UART"? 

I would have contacted you directly about the issue, but it seems impossible here to write a pm?!

kort0175, 2022-01-29:
I had to plug the USB into the ELRS module board and flash it that way.

Stekkereend, 2022-02-19:
running otx 2.3.14 on an jumper t lite and trying to flash elrs using the transmitter. As a result im getting the newest LUA is stuck in "Loading". 
baudrate option is not in the hardware tab . Also when trying to run ELRS onder tools nothing happens. When I run the lua from the scripts folder it hangs on loading

3djc, 2022-02-19:
You cannot flash elrs with OpenTX. As for baudrate, it will use 400k on tlite external

Stekkereend, 2022-02-19:
Sorry , not trying to flash Elrs via the TX but only having it recognized by the t lite. As the lua script won’t run a cannot activate the bind or the wifi update of the receiver

3djc, 2022-02-19:
Working fine
[image]

Are you using wrong script ?

Stekkereend, 2022-02-19:
Need to do factory reset off the t lite I think. When  selecting the ExpressLRS line from page 1/7 nothing happens. Also have 2 identical entries there , both not doing anything

Stekkereend, 2022-02-19:
[image]

Stekkereend, 2022-02-19:
[image]

Stekkereend, 2022-02-19:
[image]

kilrah, 2022-02-20:
Have you set the external module type to CRSF/ELRS? Won't work until that's done.

Stekkereend, 2022-02-20:
Flashed the module to v1.2.1 and now it’s has contact with the transmitter. Also both tx and rx are now visible over WiFi . Issues are only on V2.2.0 .

kilrah, 2022-02-20:
2.x uses a different set of scripts so you probably just need to load and use the correct one.