opentx/opentx #6893: 2.3 not allowing .frsk file flashing on older radio

anthsorr, 2019-10-04:
2.3 is not allowing .frsk files to be loaded on R9MM-OTA via the rear port of the Taranis X9D+SE. Can this be addressed?

3djc, 2019-10-04:
Please try 2.3.1N7 that will be available later today

closed 2019-10-04 by 3djc

fjp, 2019-11-13:
I have a similar issue with the Taranis X9D+SE using 2.3.2N41. I was able to flash the R-XSR with the latest FPort firmware (`R-XSR_FPORT_LBT_190311.frk`). However, I cannot flash the latest ACCESS firmware `RXSR-FPORT_ACCESS_191107.frsk`. 

Is this a problem with openTX? 

What is the difference between the `*.frk` file and `*.frsk` file?

bsongis, 2019-11-13:
.frsk files have a header which tells to the radio which is the destination. If your .frsk is a firmware for a receiver it won't be possible to flash it to the module by mistake

fjp, 2019-11-13:
When I try to update the firmware (`RXSR-FPORT_ACCESS_191107.frsk`) I get the following error message:

```
Firmware update error
Device not responding
[EXIT]
```

I also tried the S.Port firmware (both *.frk and *.frsk). *.frk works but I don't know how to use the *.frsk file or if it is even needed.

fjp, 2019-11-13:
> .frsk files have a header which tells to the radio which is the destination. If your .frsk is a firmware for a receiver it won't be possible to flash it to the module by mistake

@bsongis thanks for the information. Does this mean that with *.frsk files it is only possible to flash internal modules and no receivers like the R-XSR? 

I saw that .frsk firmware files should be supported in 2.3.0. 

> Support for .frsk firmware files. Those contains identification of target hardware, preventing flashing wrong device with wrong firmware type errors

Is there anything else I need to do in order to flash the .frsk firmware to the R-XSR receiver via the rear port of the Taranis X9D+SE? Thank you for your help.

kilrah, 2019-11-14:
Which X9D+ do you have? The 2019 or old version? 2019 should show the choices you have, and older "Flash external module" instead.

fjp, 2019-11-14:
I use the Taranis X9D+ 2019 SE. So it shouldn't make any difference flashing .frk or .frsk? Both should work? Thanks for the support.

kilrah, 2019-11-14:
2019 has a sport connector like the menu suggests, so you must use that, not the module pins.

fjp, 2019-11-14:
Thanks, I wasn't aware of that S.Port where the battery  is located. I will try flashing with that S.Port.

fjp, 2019-11-18:
@kilrah thank you again for the hint. Flashing the .frsk file worked using the S.Port. Is there a manual where I could've read about this S.Port? I followed too many tutorials for the X9D plus (not the 2019) version.

kilrah, 2019-11-19:
It's there, if you want to flash a device you connect it and select flash. Nothing fancy.

gnkarn, 2020-04-18:
Hi all , i have taranis XD9 plus , old model , with 2.3 firmware and trying to flash an XSR receiver , because telemetry is not working and i have no clue what firmware version is on it from factory .
I was unable to flash it using pack port, i see that mine has also a "reserved" port  , is a three pin one , more like an sport , anyone one knows if , that one is anabled as an sport , when using newer  transmitter firmware like the one i have now ? ,  thank you

kilrah, 2020-04-18:
No it's not, you have to use the module pins and the cable with swapped pins

gnkarn, 2020-04-18:
that is what i did, not working ( and works fine with other modules like Frsky X8R , any recommendations of what else to try ,  other firmware version on the taranis? , already tried with 2.1 also , tis

3djc, 2020-04-18:
If it works with X8R, it means radio+firmware+cable is ok, issue with your XSR or its connection

gnkarn, 2020-04-18:
by looking at the serigraph on the board it says X4RS rev 1.1  could it be possible that they installed the wrong firmware , I suspect of something in that line, because  radio controls work fine , then when trying to flash it makes the reset right , by then the taranis says " update error device not responding "

gnkarn, 2020-04-18:
HAVE a doubt now , not related to receiver not flashing . :  i do have connected rx6: sbus , and Tx6 to Sport , on my kakute F7 ,  is that correct ? , or should i use a different port for telemetry , than radio ? , thanks

3djc, 2020-04-19:
Your receiver must NOT be connected to anything else when flashing

gnkarn, 2020-04-19:
sure ,  my RX is not  flushing  , so gave up on that part  for the moment , need to move on .

what  i m asking is  if when using both wires ( separate rx and telemetry) ,  the connection can be on the same port (like  RX6,Tx6) , or do i need to use different ports ( RX1, TX6) , i red many cases when people with no telemetry  were moving to serial1 , for sport to work .. tks