opentx/opentx #6894: ACCESS binded RX disappear from menu if module turned off momentary

uragano47812, 2019-10-04:
**Describe the bug**
When you have already selected the protocol to use in this case ACCESS and already bind the RX in my case RX6R and want to turn off the internal module or (select another protocol) for testing propose, and then select the correct protocol again (as said ACCESS) you have the RX working back correctly but on the RX list instead of the RX name you will see [Bind] like on other 3 position. The only way to have it back is to re-do the binding to have it back on the list with it's name and to have the ability to choose the options menu like share, channels re-mapping.

 - OpenTX version 2.3.0
 - Hardware Frsky X9D+ SE 2019 with last ISRM firmware.
 - OpenTX Companion 2.3.0
#

kilrah, 2019-10-04:
Yeah, no choice. Make a copy of the model if you want to test things.

uragano47812, 2019-10-04:
ok no problem. i was testing the redundancy with an R9 rx. but no problem i will bind again.

closed 2019-10-04 by uragano47812