opentx/opentx #6897: X-Lite didn't activate Vantac MPM module

Gerd-St, 2019-10-04:
Hi,

I am using the FrSky X-Lite with official Open-TX 2.3.0 release firmware.
I was flying a quad with FrSky protocol, using the internal rf module.
After that I changed the model on X-Lite to another quad with DSMX protocol, using the Vantac MPM Lite module with its latest firmware version 1.2.1.84.
When turning on the quad it didn't connect, as if the Vantac module would not  send.
I turned off and on again the X-Lite and the quad connected normally.

This never happend with previous OpenTX versions 2.2.x.

Regards, Gerd

raphaelcoeffic, 2019-10-04:
Have you already updated the bootloader on your module? Some are known to be faulty and trouble while booting.

Gerd-St, 2019-10-04:
Yes, the module is up-to-date.

notlebowski, 2019-10-04:
You do know that the bootloader of the module can only be updated after a hardware modification and the use of an St-link.

If you update the firmware of the MPM you do not update the bootloader

Gerd-St, 2019-10-04:
You are right, I thought the bootloader would be always included. Have found the docu for updating the bootloader meanwhile and will do so if the problem will show up more often. But up to now it happend only once since I am owning both X-Lite and MPM (about a year) and so I thought it has to do with the recent Open TX 2.3.0 update.
Thank you for the hint.

closed 2019-10-04 by Gerd-St