opentx/opentx #6900: Companion Shift+Trim not working correctly

MikeNH, 2019-10-05:
*Before submitting,  search the open issue if there is already a report of the bug*

**Describe the bug**
I tried to reopen the original issue but it's not obvious to me how to do that. This is the original issue with updated information:
[github.com/opentx/opentx/issues/6582](url)  (I'm not sure why the link is not working but the screen print is attached.
[image]

**General information**
 - OpenTX version 23.0
 - Hardware X-Lite Pro
 - OpenTX Companion Windows 10

kilrah, 2019-10-05:
Working fine in 2.3.0 and 2.3.1,

closed 2019-10-05 by kilrah

MikeNH, 2019-10-05:
Thanks for the prompt response. This problem must be unique to my laptop. Feature works fine on the radio. Using the trims as switches allows for 12 channels to be in use on my quad.

Again, thanks.