opentx/opentx #6903: X9D RS not available, instead (non working) S3 shown in hardware

wilco1967, 2019-10-05:
Taranis X9D (non plus)
OpenTx 2.3.1 (5-oct-2019)

The right slider (RS) seems to be missing / not configured in the latest OpenTX for X9D

instead of RS, in radio settings / hardware menu an S3 Slider is shown, but that doesn't respond to anything.

same behaviour in the simulator, but seems correct in Companion radio settings.

483917, 2019-10-05:
Yes same for mee

Taranis X9D (non plus)
OpenTx 2.3.1 ,-haptic -flexr9m -sqt5font -lua -en

The right slider (RS) seems to be missing / not configured in the latest OpenTX for X9D

instead of RS, in radio settings / hardware menu an S3 Slider is shown, but that doesn't respond to anything.

switching back to OpenTx 2.3.0 firmware and (RS) is back

dev-fred, 2019-10-05:
Was OK in 2.3.1N2 and appears in 2.3.1N3

3djc, 2019-10-10:
Fixed already

closed 2019-10-10 by 3djc

wilco1967, 2019-10-11:
Seems to work !

Thanks !