opentx/opentx #6904: 2.3.0 / X9D - Lua script parameters incorrect after migration from 2.2.3

RC-SOAR, 2019-10-05:
Using X9D, migrating from 2.2.3 to 2.3.0 in situ (on transmitter). 

My setup uses a mixer script. It has two parameters bound to CH24 and CH25. 
After migrating from to 2.3.0, the inputs are incorrectly bound to CH23 and CH24.

[image]

RC-SOAR, 2019-10-06:
I've verified that It also fails when migrating to 2.3.1. It would great if this could be fixed in 2.3.2, as several users of my F3F template are using this script.

3djc, 2019-10-06:
Even when fixed, the fix will not apply to those who have installed 2.3.0 or 2.3.1

RC-SOAR, 2019-10-06:
Yes, understood, and thanks for fixing.

In the meantime, I'm advising users of the script to hold off migrating their 2.2 setups until the next version.

lshems, 2019-10-06:
You are all aware that different versions have different 'magic numbers'. Are all conversions covered? between different radio type's and different opentx versions?

perhaps some more elegant solution is possible??

lshems, 2019-10-06:
I know, I shouldn't suggest improvements I can't create myself. Sorry for that.

kilrah, 2019-10-06:
These are source lists, same as everywhere else in the radio, so same conversion applies. Unrelated.

3djc, 2019-10-09:
@RC-SOAR we have an issue with this. We have ne way to differentiate if the value represent a numeric value or a 'source index' (as in your case), as this is determined by the script itself, which we do not have access to. So it is impossible for us to do that migration.

RC-SOAR, 2019-10-09:
So I take it the source lists have changed in 2.3? Would it be safer to pass channel numbers to scripts rather than channel values, then use getValue()? I'd prefer to avoid that, but it might be an option if it prevents migration issues with future updates.

lshems, 2019-10-10:
Now that's a more elegant solution on the lua script side.
:)

3djc, 2019-10-10:
Yes source lists are usualy changing on major versions. in this case, a shift of 1 on x9d, 2 on x7, 4 on horus to allow new switches to be installed..

closed 2019-10-10 by 3djc