opentx/opentx #6906: Companion simulator - telemetry does not work

offer-shmuely, 2019-10-06:
telemetry does not work on simulator

**To Reproduce**
Steps to reproduce the behavior:
1. download a models with telemetry from the transmitter
2. run simulator on this model
3. enable telemetry
4. set the values in telemetry screen
5. check telemetry screen if it receive the information

**Expected behavior**
telemetry can be seen live in the telemetry secreen (the "*" andvaules should apear)

**General information**
 - OpenTX version 2.30
 - Hardware X10s
 - OpenTX Companion 2.3.0 & 2.3.1 on windows 10

[image]

Note that the ID is looking strange

[image]

kilrah, 2019-10-06:
Duplicate #6862

closed 2019-10-06 by kilrah

offer-shmuely, 2019-10-06:
Thank
Sorry for the duplication.

I think it better to label all this type of issues with “companion” label, 
I did check for same issue before i posted, but i filtered by “companion”