opentx/opentx #6908: 2.3.1 ppmus Build Option Won't Change to %

szempruw, 2019-10-06:
I installed 2.3.1 on X9D+ from 2.2.1.  I wanted to see /try ppmus box ticked to see effect. I used settings on Companion 2.3.1 and created file opentx-x9d+-lua-noheli-ppmus-sqt5font-en-2.3.1 bin fie which was downloaded in directory and subsequently written to Tx. The us was seen as expected on channel page. Previously I always used %. 

I  wanted to change back to % screen.
I created new download to directory, opentx-x9d+-lua-noheli-sqt5font-en-2.3.1 bin file, and wrote to Tx. ppmus values did not change to show %. I followed the automatic prompts to write to Tx.
Thinking wrong was inadvertently chosen by the prompt window, I repeated download and closed prompt window. I then purposely selected Icon "Write to Tx" and made sure opentx-x9d+-lua-noheli-sqt5font-en-2.3.1 bin file was selected. After writing, I checked screen and us was still present. I expected % values -100, 0, 100 to be shown and they were not.

It appears that once ppmus is chosen as a build option, it is not cleared from Tx by new bin file and it remains.

kilrah, 2019-10-06:
When you flash everything is erased and replaced with the file you flash. So either you had a wrong file, or didn't flash correctly. There's no such thing as "partial" flashing.

closed 2019-10-06 by kilrah

szempruw, 2019-10-06:
Regardless, the file opentx-x9d+-lua-noheli-sqt5font-en-2.3.1.bin without ppmus was written and I don't know what is happening except the us values remain on the channel screen. How can that be?

kilrah, 2019-10-06:
That it was not written, or that you mixed up the files while naming/saving and opentx-x9d+-lua-noheli-sqt5font-en-2.3.1.bin  actually contains opentx-x9d+-lua-noheli-ppmus-sqt5font-en-2.3.1.bin.

szempruw, 2019-10-06:
Companion downloads the files. I can easily see whether the file has "ppmus" in the file name or not. I purposefully chose opentx-x9d+-lua-noheli-sqt5font-en-2.3.1.bin file.

If you'd like, I can take step wise screen shots and tediously document what I am seeing. I worked for 10 years testing and documenting software bugs as well as URS, SRS and Test BPO's and Trace matrices for medical device software. Or you  can believe me that I am choosing the correct file, or the Companion interface is not choosing for some reason what I am. I can only report my suspicions as I did and if not good enough, perhaps give you more data.
If you have  a Taranis, you can repeat this yourself. It would be faster. Please  advise.

3djc, 2019-10-07:
Could you have a look at screen radio/version, the is  a "Firmware options" button, could you click it and check the options mentionned there ?

szempruw, 2019-10-07:
Indeed the Version check of the Firmware options says ppmus is loaded. I meticulously repeated my steps this morning at about 9:30 my time. I again checked and the ppmus is still there. I took screenshots for each step.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
Note the Companion Error message. in the 7th picture.

szempruw, 2019-10-07:
The screenshots didn't load correctly. There may be duplicates. You best look at timestamp and proper sequence. There appears to be duplicates. I can sort and resend I needed.

3djc, 2019-10-07:
Could you tick show details when flashing, because exit code 3 doesn’t sound right

kilrah, 2019-10-07:
The "could not check firmware from radio" means nothing got flashed as expected. Radio isn't found. How are you connecting it, in what state?

szempruw, 2019-10-07:
I edited the picture files.

To answer first question;. If you try to click show details, you get a ding because the error message remains. If you click the OK on the error message. Both small screens clear and you are unable to see the details. So you don't have opportunity to see details.

If I understand second question, I connect to Taranis with Taranis off with mini usb and directly to USB on PC. PC drive removal says, when connected, to STM Bootloader.

szempruw, 2019-10-07:
I searched and just found a web page from Oscar Liang who stated that on 2.3.X Nightly,, you can't write firmware to Tx using Companion. 

https://oscarliang.com/update-opentx-2-3-taranis-x9d-2019/

It appears to be on 2019 XD+ radios. I've had mine more than 2 years so I am confused. So if my problem has something to do with Companion, I will learn and try to follow Oscar's instructions and use Bootloader to solve my problem.

3djc, 2019-10-07:
Well Taranis are not flashed that way. You need to hold both horizontal trims towars center then power on. You will be in bootloader mode. Then you can connect usb and try flashing

szempruw, 2019-10-07:
Ok thank you. Sorry for this. I was confused on how to connect and it was my error.