opentx/opentx #6910: Para bluetooth upgrade

mortehoe, 2019-10-07:
I have installed ISRM upgrade module with para bluetooth module on my Frsky X10s
When I try to connect via bluetooth to my x-lite pro (also para module)I cannot connect as it seems there is no address field showing under hardware bluetooth trainer in opentx 2.3.1 (which is showing on x-lite pro opentx 2.3
I installed opentx 2.3.1 with internalaccess enabled.
When I try to discover under model setup/trainer/mode/master bluetooth the discover button does not activate and discover anything (x-lite pro set to slave/bluetooth in setup
[image]
[image]
[image]
 *Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

robustini, 2019-10-07:
Welcome to the club! :-)

robustini, 2019-10-08:
There's another issue in the assigned name of the bluetooth device.
In my Jumper T16, which mounts the X12S bluetooth module, the name change takes it, and everything is detected in lowercase even if i write everything in uppercase.
In the X10S Express instead it always remains "Hello" whatever i write, so it is OpenTX that perhaps does not send the correct instructions to the module, I cannot think that FrSky has the wrong firmware in the "PARA Wireless BT".

[image]

robustini, 2019-10-10:
@bsongis , I'm trying to compile with your fix for local tests but it gives me a lot of errors, Is it an incomplete fix or am I wrong?

[image]

kilrah, 2019-10-10:
Always include your cmake line.

robustini, 2019-10-10:
cmake -DTRANSLATIONS=IT -DPCB=X10 -DPCBREV=EXPRESS -DGVARS=YES -DDEBUG=YES ../

The problem shows up with debug enabled, should be fixed...

 if(DEBUG)
    set(BOOTLOADER_SRC
      ${BOOTLOADER_SRC}
      ../../../../../${STM32LIB_DIR}/STM32F4xx_StdPeriph_Driver/src/stm32f4xx_usart.c
      ../../../../../serial.cpp
      ../aux_serial_driver.cpp
      )
  endif()

DEBUG should be fixed....

closed 2019-10-10 by bsongis

robustini, 2019-10-10:
@bsongis please can you merge this fix in the 2.3 branch?

3djc, 2019-10-10:
Did you even try again before posting that ?

robustini, 2019-10-10:
If you mean 2.3 branch I compile it without problems.

mortehoe, 2019-10-10:
Will this be in the 2.3.2 release.I cannot see this under issues, pull request or closed.

robustini, 2019-10-11:
> 
> 
> Will this be in the 2.3.2 release.I cannot see this under issues, pull request or closed.will this work for the X10S para upgraded module

I've tried it locally with your own tx and now it works, obviously it's the developers who decide when to merge that fix.

mortehoe, 2019-10-11:
Okay thank you.

3djc, 2019-10-11:
It has been merged this morning, and will be available in 2.3.2