opentx/opentx #6911: Para bluetooth module for X10S

mortehoe, 2019-10-07:
I have installed the upgrade ISRM module kit and Para bluetooth module on Frsky X10S.
Unfortunately the  Bluetooth setup does not show an address field (as shown in my x-lite pro)
When I try to connect "discover" from bluetooth/master X10 to bluetooth/slave x-lite pro it does nothing when clicked.
If I enable telemetry then I can see the bluetooth module on my phone so it is transmitting something.
[image]
[image]
[image]

kilrah, 2019-10-07:
Duplicate #6910

closed 2019-10-07 by kilrah

robustini, 2019-10-07:
Welcome to the club!