opentx/opentx #6912: Under X9D models with XJTD8 in .otx becomes XJTD16 in radio

dev-fred, 2019-10-08:
**Describe the bug**
In 2.3.1  after a transfer of .otx file in a X9D, all models using a XJTD8  internal are converted to XJTD16

**To Reproduce**
1-From companion open a model with internal RF XJTD8
2. run the simu you will see D16 

[image]

dev-fred, 2019-10-08:
It's fixed with the last correction of RS problem in X9D, I still have the problem in simu because the dll is not yet updated, works fine with radio

dev-fred, 2019-10-08:
In fact the problem can be reproduced directly in companion and it's not link to the firmware or the simulator.
To reproduce: 
1- create a model with XJT D8 internal
2- save it
3-close the otx windows
4- open the otx file
5- the HF internal is now in XJT D16

dev-fred, 2019-10-08:
It's really an issue, sorry

dev-fred, 2019-10-08:
Idem on X9D+, X10 so it's not only X9D

notlebowski, 2019-10-08:
Indeed... when did that happen?
I don't think it was in the 2.3.0 nightlies...

3djc, 2019-10-09:
At least 2 monthes ago, according to git blame

PR #6919 (2019-10-10): Fixed module data type export
This fixes #6912

Fixes the multi-protocol and ISRM subType as well.
closed 2019-10-10 by bsongis