opentx/opentx #6921: X9D-2.3.1 INPUTS/MIXER missing "source" mapping 3 position switches from SA - SH

astray1988, 2019-10-10:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
I have upgraded my Taranis X9D plus to 2.3.1 using Companion. When I create new quadcopter model, it's missing the 3 position switches bindings from the "Source" under the MIXER section
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'MIXER' using ‘PAGE’ BUTTON
2. Click on one of Channel and open 'EDIT'
3. Navigate to 'Source' section.
4. I don't see 3 position switches from 'SA' to 'SH' option

**Expected behavior**
I should be able to bind the 3 position switch, 

**Screenshots**

**General information**
 - OpenTX version (2.3.1)
 - Hardware (Frsky X9D+)
 - OpenTX Companion (2.3.1)

astray1988, 2019-10-10:
I also load my model to the Companion, I don't see the 3 position switches appear in the Source

3djc, 2019-10-10:
[image]

All here

astray1988, 2019-10-10:
I just figured out by myself. I need to go to the HARDWARE section to redefine all the switches, afterwards making all the changes. The SA - SH switches will appear under the Source.

closed 2019-10-10 by astray1988