opentx/opentx #6928: Use Crowding for translations

McGiverGim, 2019-10-10:
You have a good tutorial for the people that want to help with translations. But the translations can be done by people who knows nothing about computer things.

This is only a suggestion: have you tried to upload your ts files to a translators portal like Crowdin? Is free for Open Source projects, it seems accept the TS files and in this way A LOT of people can help with it.

For this to work you will need:
- Automate in some way the upload of the translations to the portal.
- If you want, someone must approve them, so you will need volunteers to proofreading each language.
- Someone must incorporate them to the code before the release, if you don't automate it.

But if you want to test it, you can simply do it in a manual way to see how it works before automate nothing.

I can talk about the Betaflight Configurator app as a experience with Crowdin. The translators come and go, but 250 people have helped at some moment with it until now.

Eldenroot, 2019-10-10:
+1 great idea!

comradekingu, 2020-05-07:
@ McGiverGim @Eldenroot Localization is a great idea, but Crowdin isn't.
If you read its terms and conditions, you will find tracked profiling, and all the venues of selling data to advertisers and data collection agencies.
It isn't gratis for libre software either.
> 
>     Crowdin for Open Source
> 
>     We support the open source community. If you're building awesome non-profit projects that could use the power of Crowdin, we're happy to help. Make your content multilingual to reach a global audience.
> 
>     Apply for a free open source license

What that means is having a single source of income, donations or otherwise, prevents a project from qualifying for gratis hosting. Furthermore, reading the terms and conditions, doing so means the project has to contribute to their closed translation memory. Effectively dual-licensing the string-base. "Open Source for the closed software Crowdin" would be a more apt description.

Furthermore it isn't a good tool, all of which pales in comparison to the libre software Weblate.

https://hosted.weblate.org/projects/xcsoar/ is already on Hosted Weblate btw.
Happy to contribute to both as long as I can do so without supporting the likes of Crowdin :)

McGiverGim, 2020-05-07:
Sell the data to advertisers or similar is not a problem: the translations are public, in or outside Crowdin, so they can get the data from GitHub if they want.

I'm not too sure of the interpretation. Non profit does not imply not having donations. It imply the project/developers not earning money with it in any way. If you receive donations but are for maintenance of the project, hosting, etc. my understanding is that you can apply for the license. There are lots of open source projects that accept donations and use the free license of Crowdin. 

But this is not the theme. The tool is indifferent if it does the job. I like Crowdin but if the tool is other for this or other reasons, I will help too. I have used several translations portals in the latest years. Not problem at all.

comradekingu, 2020-05-07:
@McGiverGim That isn't the data they use. No, non-commercial covers donations and any other source of income. Even if you employ your own interpretation, it would artificially limit donations to the level of upkeep, which still doesn't make any sense.
With that out of the way, yes. Weblate is a far superior tool.

closed 2021-02-04 by stale[bot]