opentx/opentx #6933: Bring back RAS/SWR to X9D+! Please!

throttlenerd, 2019-10-11:
Updated my X9D+ (pre 2019) to 2.3.1, "noras" option unchecked, but now I don't have the RAS/SWR readings anyway! How the føk should I know if I attach a good antenna and which way to bend it the best? Did a bit of a research before posting but couldn't find anything rather than a request to bring down the RAS reading feature, eh! I want the option for my SWR to be on the screen all the time, like battery voltage or current time )

Thanks!

notlebowski, 2019-10-11:
The "noras" option is intended for the first X9D+ models. I have one. I believe it was released in 2016.
These models don't have working hardware to provide RAS. 
These X9D+ models (I believe it's A01) should have "noras" checked.
You're saying you have a pre-2019 model.
That info is not relevant. 

My X9D+ will never output a sensible RAS-reading. If yours is the same model it will be the same.

kilrah, 2019-10-11:
https://github.com/opentx/opentx/issues/6775

closed 2019-10-11 by kilrah

throttlenerd, 2019-10-11:
> My X9D+ will never output a sensible RAS-reading. If yours is the same model it will be the same.

I don't know which exact version my X9D+ is, but yes, it's a few years old, so "Pre 2019" is the only correct description I can tell. And it showed the RAS correctly! I mean, RAS of 0-1 is better than 10, right? Or RAS of 10 will not affect the range? And why "noras", why even bother deleting the feature which is not that simple to get into? It involved a few long-presses, after all!)

kilrah, 2019-10-11:
> I mean, RAS of 0-1 is better than 10, right?

Not significative.

> And why "noras"

Becasue the first batch of X9D+ modules give a RAS value but it's invalid.

notlebowski, 2019-10-12:
If the X9D+ just came out when you bought it and stock was low at the time, the chances are high you have faulty hardware that's unable to tell you if there's something wrong with your antenna.

Even if RAS is working as it is supposed to, its significance is overstated by some. 

People with intentions to improve their range by modifying their setup by replacing the fixed antenna with a removable one often achieve the opposite. 

The end stage of the RF module is highly sensitive to ESD and can be damaged by mere touching it. Often this damage is partial and deteriorates slowly over time. This is treacherous and it also gave those "pioneers" the time to post YouTube videos of their "successful upgrade"