opentx/opentx #6935: Widgets using source decrease by 2

MRC3742, 2019-10-12:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

kilrah, 2019-10-12:
Going to 2.3.2 from any JumperTX version will not be supported and disabled.

closed 2019-10-12 by kilrah

MRC3742, 2019-10-12:
I knew that OTX was supporting T16 on newer 2.3 versions and thought that you would be looking for feedback on upgrading. I didn't realize that JumperTX 2.2.3 was being considered an unsupported version but fully understand your position. Thanks for all your work on this project.
Rich