opentx/opentx #6938: R9M 2019 Access is missing on Horus x10s with ISRM upgrade

Nico9n, 2019-10-12:
I flashed my Horus x10s with internal acceess upgrade. I think now it should be possible to use the external R9M 2019 access modul. The modul is missing in configuration on the transmitter. In Companion it is selectable. When I save the configuration to the transmitter, the external module is off.

3djc, 2019-10-12:
No, it doens't work like that. In order to have R9M 2019 ACCESS working, you need an high speed link to this module, which only exist in the new radiothat natively support ACCESS.

Changing your internal module hasn't changed to link to the external module, it cannot work on X10 non express, wheter or not you have an upgraded internal module

closed 2019-10-12 by 3djc

Nico9n, 2019-10-12:
I have asked Frsky about that 2 Months ago. They said R9M 2019 should should work with access in the old Horus.

kilrah, 2019-10-12:
They have recently been telling people the opposite, so maybe they changed plans.

Nico9n, 2019-10-14:
I asked Frsky again. It should work.

3djc, 2019-10-14:
Could you paste the question and their response please ?

Nico9n, 2019-10-14:
From Frsky:
> Hello sir, in order to have the R9M 2019 work with ACCESS firmware with older Horus X10S, you need to purchase an upgrade internal Tx module to replace the original one. Here is the purchase link from our dealer: https://www.horusrc.com/en/frsky-horus-x10-isrm-rf-module.html

I can send you a screenshot If you want.

3djc, 2019-10-14:
Yes please, we will double check with them, because changing internal module does not change the hardware link to external module, incapable of doing the required 450kbps required by R9M 2019 ACCESS

Nico9n, 2019-10-14:
https://1drv.ms/u/s!AiziGjacV8eZgtMmWU7iJt7SqsAwYQ here it is

raphaelcoeffic, 2019-10-14:
@Nico9n well, if they explain us how this is supposed to work, why not! But for now, we know only about the serial interface, and there is no way we can do software serial with that speed....

bsongis, 2019-10-14:
Confirmed, it won't work

Rogero-, 2019-10-14:
if ACCESS doesn't work with the upgrade module they are selling, the question is, why the upgrade module was called: "FrSky ACCESS Upgrade ISRM RF Module for Horus X10/X10S" and what does it upgrade exactly if ACCESS still won't work, arghhh FRsky what a mess.....

raphaelcoeffic, 2019-10-14:
Well, you get ACCESS on the internal module on 2.4Ghz. Just not on 900Mhz with the external module, which, by the way, has no synchronisation and thus, quite some sampling aliasing on the receiver side. Ask the BF guys for it, they will tell you some stuff about it ;-)

Rogero-, 2019-10-14:
thanks raphaelcoeffic for clearing that up... i'm glad i ordered a Jumper T16 Pro and not another Frsky radio.