opentx/opentx #6939: T12 Internal Multiprotocol module support

longxk, 2019-10-12:
The T12 transmitter has a empty circuit  inside for internal module, and it is connected to the main board with a 12 pin connector. pin 1-9 are used for usb and trainer port, the additional pin 10-12 are not used for now. **pin 11** is connected with JR module bay's signal pin, **pin 12** is connected with JR module bay's telemetry pin, so **pin 10** is very likely for power.

So it seems like adding a internal multiprotocol module in hardware is fairly easy. Just strip down the external module case, put it into the transmitter, wire it to internal pins.

But opentx firmware disabled internal module support for T12. Since the T16 has internal multiprotocol module support already, maybe T12 can have it too? I will be glad to do some test for you guys.

[image]

raphaelcoeffic, 2019-10-13:
The issue is: at least some devs should have it. As far as I know, there are no official plans for such a module. Coding blindly could be a very tedious process I’m afraid.

3djc, 2019-10-13:
Actually fitting a full size multi board in T12 won't fit, there is simply no room for that I believe. The proper solution would be to populate the internal module board that you picture, but sourcing the parts  is not that easy, and thats not an easy soldering job either. Or maybe as Jumper to release one

PR #7072 (2019-11-21): Prepare T12 internal multi handling
This fixes #6939
closed 2019-11-21 by 3djc