opentx/opentx #6940: Trainer mode wont stay in MASTER/SBUS

jims1234, 2019-10-13:
*Before submitting,  search the open issue if there is already a report of the bug*// I went back a few pages and could not find one similar to this//

Also often minor bugs are already fixed in nightly builds,// I haven't tried any nightly builds so it may be resolved, I'm not sure// 

**Describe the bug**
//See RCGROUPS Taranis: How To thread post #6939 33512. Basically you can set the Trainer mode to Master/SBUS but it reverts back to MASTER/JACK when you either read or write models and settings" using Companion 2.3.1. Without this, a SBUS receiver like an XM+ installed in the MASTER Radio fails to link with a nearby transmitter. Writing a model with MASTER/SBUS set also reverts back.    

**To Reproduce**
Steps to reproduce the behavior:
1. Turn On Student Radio and select the model for Trainer Mode with the XM+ Rx mounted in teh Master's JR BAY. Go to 'Model settings page 2/12 set Trainer mode in XJT-D16 to MASTER/SBUS. 
2. Note all works fine. but then save this with a "Read models and settings" into Companion 2.3.1 and write it back with a Write Models and Settings.. and you will find the trainer mode setting gas reverted to the default MASTER /JACK again.      
3. The model and the XM+ are both Bound properly but unless the XM+ receiver is used in MASTER /SBUS it can'r go green or pass any CH data from the students radio 

**Expected behavior**
// When a user reads settings from a radio they shouldn't revert to a default.. When a user writes them back into the Radio after changing them again.. it shouldn't revert back to a default Trainer mode. Occasionally some other settings have changed too (like the Receiver No. once  // 

**General information**
seeing this behavior with 4 of 4 different models used in training mode with FrSky RX and GRX receivers. 
 - OpenTX version (e.g. 2.3.1 in both Radios)
 - Hardware (e.g. Frsky Q7XS in both radios running latest firmware and bootloader 2.3.1)
 - OpenTX Companion (2.3.1)

kilrah, 2019-10-13:
You don't say what radio.

3djc, 2019-10-13:
I can replicate on X7

closed 2019-10-13 by kilrah

jims1234, 2019-10-13:
kilrah: note issue states" Hardware (e.g. Frsky Q7XS in both radios running latest firmware and bootloader 2.3.1)" Glad 3djc was able to replicate it and that this issue is now closed but will be addressed in issue Trainer mode handling #6943.. wow that was fast and efficient ! Thanks!

3djc, 2019-10-13:
It will be in 2.3.2 when that becomes available

jims1234, 2019-10-13:
Awesome! In the meantime I can just check and set it manually in the  x7 MASTER radio.. thanks so much for doing what y’all do back there. It’s pretty terrific , not only what, but how you do it 

Thanks!

PR #6943 (2019-10-13): Trainer mode handling
This fixes #6940