opentx/opentx #6945: Persistent timer data displays incorrectly on QX7 after upgrade to 2.3.1

VancouverUmbrella, 2019-10-13:
After updating my QX7 from OTX 2.2.4 to 2.3.1, the persisted timer data for each of my models displays incorrectly on the screen of the radio. 

For example, instead of 64:46:46 (correct value), the radio displays 3886:4. 

The problem also occurs in the Companion simulator. However, in the Companion editor interface, the data displays correctly.

Edit UI. See persistent value at far right of screenshot.
[image]

Simulator UI. The same occurs on the actual radio screen.
[image]

Radio: QX7
OS: macOS Catalina 10.15

VancouverUmbrella, 2019-10-13:
Is the problem that, on the radio, both the display and edit interfaces no longer have an hours field and are restricted to minutes and seconds i.e. MMMM.SS not HHH.MM.SS.

8504 minutes = 141.73 hours.

3djc, 2019-10-14:
That, and the new TIMEHOUR format is HH.MM.SS, which cannot fit on the 128x64 screen, so we would need to restrict to HH:MM, but then some will complain because if they use a more traditional timer, they want MM:SS. But anyway in your case, 141h44 would not fit on screen either

VancouverUmbrella, 2019-10-14:
Hey 3djc, thanks for the quick response!

Perhaps the time format could automatically adjust to optimize the data as MM:SS or HHH:MM or simply HHH.H (as a decimal) as required? In the current implementation, which renders as MMMM:S[S] for all my models, the last second digit is always occluded. 

Plus, anything over 60 minutes is pretty hard to digest without a calculator -- for me anyway. BTW I never noticed a problem with the display before. Perhaps the last digit was cut off? I don't recall. :-)

As a glider pilot, I tend to put a lot of time on all my models. However, I expect it'll be years before I exceed 999.59 hours on a single model.

Love your good work. Otx and Companion are excellent.

All the best,
Andrew

3djc, 2019-10-18:
Given the space on screen, the limit would be 99h59mins. if you pass that we could do HHHHh, I do not think HHH.H would be valid, would 100,5h be 100h30min ? 100h50mins ? very confusing

VancouverUmbrella, 2019-10-18:
Your suggestion makes sense to me i.e. MM:SS -> HH:MM -> HHHH as required.

FWIW in the old scheme things weren't bad. One or perhaps two numbers could get partially or fully cut off, but I had no trouble understanding a log time of HHH:MM e.g. 141:33.

Thanks for considering this.

PR #7022 (2019-11-04): Rework 128 timer to better leverage tiny space
Some users display persistent timers on main view, leading to issue given the tiny space allowed.

This change will display MM:SS for timer under 1 hour

It will then display HH:MM for timer lower than 99h59

Then just the number of hours for timers above that

This fixes #6945
closed 2019-11-04 by bsongis