opentx/opentx #6946: Setting the date and time needs to be done twice

mpjf01, 2019-10-14:
When the RTC battery goes flat and gets replaced it is necessary to reset the date and time. After this data is set correctly and the radio is power cycled the date is set to 2000-01-10 and the time 00.00.00 regardless of the user setting. After setting the date and time correctly for the second time, following a power cycle it appears correctly.

3djc, 2019-10-14:
Radio ?

Have you waied enough on your first try to allow rtc to be set  ?

mpjf01, 2019-10-14:
X9D and X12S. How long should I expect to wait?

3djc, 2019-10-14:
2.3.1 ?

RipperDrone, 2020-04-11:
Issue confirmed, on X9D+ (2013 model). After RTC battery replacement, year shown was 1970 (?), then after setting date for 1st time, it didn't stick except the year changed to 2000 (pls don't ask me why :-)), then after 2nd attempt setting date and time correctly, it DID stick. No idea what's going on...

closed 2021-01-09 by stale[bot]

RipperDrone, 2021-01-09:
since no fix has been proposed, should we keep it from being auto-closed?