opentx/opentx #6953: Pre-Built nightly companions don't include the T16

tenten8401, 2019-10-14:
Downloading a fresh 2.3 companion nightly from the website doesn't seem to be built with Jumper T16 support enabled.

I'm downloading this:
https://downloads.open-tx.org/2.3/nightlies/companion/linux/companion23_2.3.1N17_amd64.deb

When I run it I'm not seeing an option for the T16 at all under my radio profile:
[image]

I'm not sure if it's a missing build flag on the server making the nightlies or if it's something I'm doing wrong, but building it myself on a local server using the docker image does have the "T16 / T16+" option in there:
[image]

3djc, 2019-10-14:
When we start 2.3.2 nightlies, it will be available

closed 2019-10-14 by 3djc

tenten8401, 2019-10-14:
Ahh that makes sense.