opentx/opentx #6954: OpenTX 2.3.1 X9E bugs

PascalRZ, 2019-10-15:
First of all, congratulations for your job!

Problems on my X9E, coming from 2.1.9 to 2.3.1, because frsky XM+ can’t be bind with an older version of X9E (there is no information about this problem on frsky website. This is not due to LBT or non-LBT).
1°) I first try to upgrade from 2.1.9 to 2.3.1 directly: all my models were unusable, outputs, variables, etc… were lost. Another problem was a new EEPROM conversion each time I started the transmitter. Opentx 2.3.1 badly convert an 217 EEPROM to a 219 EEPROM and do not save it.
2°) Found on forums, I must upgrade from 2.1.9 to 2.2.4, and then from 2.2.4 to 2.3.1 : OK, it works now, my models have been well converted and EEPROM was converted only one time.
1°) & 2°) Can you add an error message saying an intermediate upgrade in 2.2.x must be done ?

3°) Now, my X9E in opentx 2.3.1 has lost the two lateral pots RS and LS. There are missing in the calibration menu. There are only 4 pots instead of the 6 pots delivered with the X9E. I added its manually but can you correct this bug in the X9E native configuration ?

PascalRZ (France)

3djc, 2019-10-15:
Please check your hardware settings page, I bet the missing pots are simply not declared

3djc, 2019-10-16:
I believe it is not, if you eeprom reset in 2.3.1, you have all your inputs ok I think. We have seen rare occurence of what your experience in 2.2.x, so likely there is some rarely occuring issue in the 2.1 to 2.2 process

PascalRZ, 2019-10-27:
It was just a warning about bugs I found during my update form 2.1.9 to 2.2.4 to 2.3.1. If you really think these problems will not occur to other X9E users, I close the issue. If somebody gets the same problem, he will reopen the issue.

closed 2019-10-27 by PascalRZ

lepicfoto, 2019-10-28:
hi to all :-)
my experience with my beloved X9E. I use 2 as a instructor / student set.

Always start with a switch warning, always no matter how I set my warnings or no warnings at all. When I do not use a switch I get off to a good start, there is just the small problem when I turn on the ignition to my engine Or use any switch at all, then I immediately get a switch warning, it actually do not matter what switch I enable = immediate warning. So I can't get past this little hmmm upset :-)
I started with a new receiver ((which worked perfectly on everything up to and including 2.3.0.) It also works completely flawlessly on my X9D 2.3.0 And my student X9E 2.3.0 ) I added the things I need, flameout switch, servos, etc. But i can't get past having to use a switch, an.y switch, i have tried 5 different switches = immediate switch warning. , anyhow just my little owwww it's not good not good, at all.
Sorry it got a little long, sorry..