opentx/opentx #6956: Connect to PC over bluetooth instead of USB

zwn, 2019-10-15:
It would be great if bluetooth could be used for more stuff. I'd love to see my X-Lite S as bluetooth joystick (just like over usb). Also connecting to the companion through bluetooth would be great.

3djc, 2019-10-15:
Yes it would be great, but BT in those radio is BLE only and cannot beused frot hings like the ones you mention

zwn, 2019-10-15:
https://en.wikipedia.org/wiki/Bluetooth_Low_Energy suggests that HID is possible using BLE. Also sending custom data is supposed to be easier with BLE than with Bluetooth Classic, which could be used to communicate with the companion. For working with the eeprom not much bandwidth would be needed. What makes you convinced BLE cannot be used the things I've suggested?

robustini, 2019-10-15:
It seems to me a fantastic idea, to be considered absolutely, also for the joystick option, BLE would succeed in both cases, in the second it already does so with Bluetooth trainer.

jace26, 2020-02-13:
+1
This seems to be a no brainer. If it's true that BLE can be used in this way then this would be a selling point for the radio and it seems absurd to have bluetooth and not have it be useable as a joystick. 

Is Frsky just trying to sell dongles? Why isn't this a native feature?

kilrah, 2020-02-14:
The achievable update rate over BT would likely be unsatisfactory especially for those wanting low latency just like it is for the trainer function, too slow.

zwn, 2020-02-14:
> The achievable update rate over BT would likely be unsatisfactory especially for those wanting low latency just like it is for the trainer function, too slow.

That just seems like a hand-waving. There are [bluetooth joysticks that people consider][1] good wrt. latency. As for what latency is considered ok for gaming see [reddit][2] or [wikipedia][3] for some opinions.

[1]: https://www.reddit.com/r/RocketLeague/comments/8wvf9u/controller_input_lag_comparison_more_info_in_the/
[2]: https://www.reddit.com/r/xboxone/comments/7esjn9/how_much_is_too_much_input_lag/
[3]: https://en.wikipedia.org/wiki/Input_lag

jace26, 2020-02-14:
Not everyone needs low latency, supposing that is even a problem, for example, I use these controllers in a professional capacity flying large aircraft which I sometimes simulate on PC. Latency isn't an issue for these aircraft real world or in a sim because they're so slow. .

kilrah, 2020-02-15:
Making a good USB joystick is totally possible yes, but hackishly emulating one with the firmware that is in FrSky's BT modules that is not made for it in any way and is notoriously slow is something else.

markran, 2020-05-19:
+1 on using the Bluetooth for more things than Trainer such as joystick. Would be nice not to have to need a dongle.

The other thing that would be neat is if it could send voice prompts to a bluetooth headset. I think this requires an A2DP stack be implemented (if not already available for that chipset). https://electronics.stackexchange.com/questions/100786/does-bluetooth-le-stack-supports-a2dp-profile

That thread says it may not be efficient to stream audio for long periods but the RC application of occasional short prompts would be quite different than the normal 'listen to music/podcast' or telephony use cases. It also says in that thread that some Bluetooth chipsets have both normal and BLE modes. Since our radios have big batteries (compared to a headset or car key), hopefully manufacturers included dual mode chipsets.

digaus, 2020-06-24:
Adding Bluetooth support for connecting to companion would be very nice.

closed 2021-03-26 by stale[bot]