opentx/opentx #6957: Lua Display Functions: Behaviour of Text Attributes has changed since 2.3

Matze-Jung, 2019-10-15:
The code ..
```
function test(value, format)
    lcd.clear()

    lcd.drawText(10, 1, value, format)
    lcd.drawText(lcd.getLastPos(), 1, "V", format)

    lcd.drawNumber(10, 15, value, format)
    lcd.drawText(lcd.getLastPos(), 15, "V", format)
end

return {
    run=function()
        test(80, MIDSIZE+PREC1)
    end
}

```
.. outputs:

opentx-companion-2.2.4-lua-eu_X7
[image]

opentx-companion-2.3.1-lua-eu_X7
[image]

Is there a solution I do not see, and is the current behavior of text attributes intended?
Thanks

bsongis, 2019-11-15:
PREC1 cannot be passed to drawText!

closed 2019-11-15 by bsongis

Matze-Jung, 2019-11-15:
@bsongis It was possible and totally fine on OpenTX < 2.3.x and the new behavior is not satisfactory at all. It seems to mess up with the _lcd.getLastPos()_ function. May have a look at a real use case here: [opentx-lua-widgets](https://github.com/Matze-Jung/opentx-lua-widgets/blob/master/src/SCRIPTS/WIDGETS/value.lua).

bsongis, 2019-11-15:
again PREC1 is not compatible with drawText (unpredictable behavior)
it should only be used with drawNumber

Matze-Jung, 2019-11-15:
Ok thx, I'll deal with it. But it's a pain.

bsongis, 2019-11-15:
you have to understand that we added new flags such as `CENTERED` in 2.3 and we don't have enough space to have different flags values for text and numbers.

you can use a bitwise operator to avoid passing PREC1 to drawText function, something like `(flags & (~PREC1))`

Matze-Jung, 2019-11-15:
That seems to be the hint I was looking for 👍